FastLED/FastLED #808: Using CRGBSet to define multiple ranges

marcdale99, 2019-05-27:
Hi,

I'm using CRGBSet successfully where all I need is the ability to set a single range of pixels. In this example I split my whole led strip into 4 consecutive ranges so I can light one window at a time.

CRGB leds[NUM_LEDS];
CRGBSet AllWin(leds, NUM_LEDS);
CRGBSet **Win1**(AllWin(0,11));  //Window 1
CRGBSet Win2(AllWin(12,23)); //Window 2
CRGBSet Win3(AllWin(24,35)); //Window 3
CRGBSet Win4(AllWin(36,47)); //Window 4

fill_gradient_RGB(**Win1**, 0, clr2, loc1, clr1);

Will light up a whole window (**Win1**), no problem at all.

The problem comes when I want to CRGBSet 'multiple ranges' in one set.

So, my issue is that I want to light up the 'outside edge only' of all windows at the same time. I want to use the fill_gradient_RGB tool in FastLED. For that I need a defined CRGBSet.

CRGBSet AllOuter(leds, NUM_LEDS);
CRGBSet **Win1Outer**(AllOuter(3,8));  //Window 1 Outer Edge
CRGBSet Win2Outer(AllOuter(15,20)); //Window 2 Outer Edge
CRGBSet Win3Outer(AllOuter(27,32)); //Window 3 Outer Edge
CRGBSet Win4Outer(AllOuter(39,44)); //Window 4 Outer Edge

fill_gradient_RGB(**Win1Outer**, 0, clr2, loc1, clr1);

Will light up the outer edge of **Win1**(**Win1Outer**), no problem at all.

But how do I define ranges (3,8)+(15,20)+(27,32)+(39,44) in one CRGBSet instead of 4 separate ranges? I need the fill_gradient_rgb tool to see one CRGBset so it handles it as one range of pixels.

Thanks

focalintent, 2019-05-28:
CRGBSet doesn't support disconnected ranges yet - it's on the list of things that I'd like to do with it, but haven't had a chance to yet.

marcdale99, 2019-05-28:
Thanks for taking the time to reply. I thought that might be the case. 

Late last night I had a moment of clarity and realised that in my scenario I was able to get the effect that I wanted by just addressing the whole string with my effects but also setting the CRBGSet that I didn't want lit up to be ::Black before I called FastLED.show.

It works great. I still get the one continuous effect from end to end but only showing in the places I want it to.

Thanks again