FastLED/FastLED #809: PINs do not work on STM32F103

gitmek, 2019-05-30:
@kolyshkin 
"I generally tell folks to use integer pins"   I tried to use integers like 10 for PA0 ,   11 for PA1 , then it compiles but the application does not seem to run on microcontroller (crashes ( hangs at FastLED.addLeds)  (I see it on UART activity)
then I tried   14,15,16,17 for PA4,5,6,7  then it runs, but the led strips does not do anything

as for the compiler flags, I hope that becomes a permanent fix at some point, or easily selectable.

_Originally posted by @AndKe in https://github.com/FastLED/FastLED/issues/741#issuecomment-466450888_

sallevan, 2019-08-24:
And without integer notation (i.e, using PA0/PA1/etc) it does not work ether, and I also prove this to myself using oscilloscope. :)

burner-, 2023-06-25:
I like to bump this problem. 4 year later and this problem is still here.