FastLED/FastLED #810: need  code support for color palette RGB

dursununal, 2019-06-01:
need  code support for color palette RGB

 1 buttons (Pin2 input)

 static uint8_t startIndex = 0;
 startIndex = startIndex += 1;  // index(1 to16) (Pin2 input) changes

 2 buttons (Pin3 input)

  for ( int i = 0; i < NUM_LEDS; i++) {
  leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness,    currentBlending);
   colorIndex += 1; // index( 1 to 16 ) (Pin3 input) changes

If 2 buttons with indexes I can changes, I can do 50 different animations
found sample code on the internet,but it was incompatible with your code
 
internet sample code such
 if(selectedEffect>18) { 
 selectedEffect=0;
  EEPROM.put(0,0);
switch(selectedEffect )  // case options case 1 to 18

I want help from you
thanks in advance
Dursun unal

focalintent, 2019-06-06:
For general “help with code” questions like this - you will probably get more responses in the reddit group - http://fastled.io/r - we try to keep the issues here in github focused on library bugs/issues.  (Also, there are significantly more people that read/follow the reddit group than the github issues here)

dursununal, 2019-06-07:
ok thanks