FastLED/FastLED #811: Error compiler ESP32 V 3.2.6 and 3.2.7

copilador, 2019-06-07:
Hi.
I get an error when compiling for an ESP32 (DOIT ESP32 DEVKIT V1, 80Mhz), with the latest versions, with for example version 3.2.5 everything is correct, for example i try with the example Blink and I get the following error:
Using FastLED library with version 3.2.7 in the folder: /home/co/Arduino/libraries/FastLED 
exit status 1
Error compiling for DOIT card ESP32 DEVKIT V1.

Thanks

focalintent, 2019-06-07:
I’d need to see the entire build output

copilador, 2019-06-07:
HI.

Arduino:1.8.9 (Linux), Tarjeta:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

/home/co/arduino-1.8.9/arduino-builder -dump-prefs -logger=machine -hardware /home/co/arduino-1.8.9/hardware -hardware /home/co/.arduino15/packages -hardware /home/co/Arduino/hardware -tools /home/co/arduino-1.8.9/tools-builder -tools /home/co/arduino-1.8.9/hardware/tools/avr -tools /home/co/.arduino15/packages -built-in-libraries /home/co/arduino-1.8.9/libraries -libraries /home/co/Arduino/libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path /tmp/arduino_build_917952 -warnings=default -build-cache /tmp/arduino_cache_647628 -prefs=build.warn_data_percentage=75 -verbose /home/co/Arduino/libraries/FastLED/examples/Blink/Blink.ino
/home/co/arduino-1.8.9/arduino-builder -compile -logger=machine -hardware /home/co/arduino-1.8.9/hardware -hardware /home/co/.arduino15/packages -hardware /home/co/Arduino/hardware -tools /home/co/arduino-1.8.9/tools-builder -tools /home/co/arduino-1.8.9/hardware/tools/avr -tools /home/co/.arduino15/packages -built-in-libraries /home/co/arduino-1.8.9/libraries -libraries /home/co/Arduino/libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path /tmp/arduino_build_917952 -warnings=default -build-cache /tmp/arduino_cache_647628 -prefs=build.warn_data_percentage=75 -verbose /home/co/Arduino/libraries/FastLED/examples/Blink/Blink.ino
Using board 'esp32doit-devkit-v1' from platform in folder: /home/co/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/co/Arduino/hardware/espressif/esp32
Detecting libraries used...
/home/co/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/config -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid/api -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/bt -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/driver -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32 -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_http_client -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/esp-tls -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/heap -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/log -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/smartconfig_ack -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/coap -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/console -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/expat -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/json -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/soc -I/home/co/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10809 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -I/home/co/Arduino/hardware/espressif/esp32/cores/esp32 -I/home/co/Arduino/hardware/espre [... cut ...]
/home/co/Arduino/libraries/FastLED/FastLED.h:355:25: note:   template argument deduction/substitution failed:
/home/co/Arduino/libraries/FastLED/FastLED.h:315:56: note:   cannot convert '& c' (type 'NEOPIXEL<3u>*') to type 'CRGB*'
   return addLeds(&c, data, nLedsOrOffset, nLedsIfOffset);
                                                        ^
Usando librería FastLED con versión 3.2.7 en la carpeta: /home/co/Arduino/libraries/FastLED 
exit status 1
Error compilando para la tarjeta DOIT ESP32 DEVKIT V1.

commit cf8ce865 (2019-06-07): Fix #811 and #812 - took a bad PR that broke the build for other platforms
closed 2019-06-07 by focalintent