FastLED/FastLED #812: Cannot get blink to compile on arduino 1.8.9 mac and latest core for esp8266

delhoume, 2019-06-07:
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/delhoume/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/delhoume/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/delhoume/Documents/Arduino/libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10809 -build-path /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693 -warnings=default -build-cache /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_cache_884372 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -verbose /Users/delhoume/Documents/Arduino/libraries/FastLED/examples/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/delhoume/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/delhoume/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/delhoume/Documents/Arduino/libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10809 -build-path /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693 -warnings=default -build-cache /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_cache_884372 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -verbose /Users/delhoume/Documents/Arduino/libraries/FastLED/examples/Blink/Blink.ino
Using board 'd1_mini' from platform in folder: /Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2
Using core 'esp8266' from platform in folder: /Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2
Detecting libraries used...
/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/d1_mini /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/sketch/Blink.ino.cpp -o /dev/null
/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/d1_mini -I/Users/delhoume/Documents/Arduino/libraries/FastLED /var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/sketch/Blink.ino.cpp -o /dev/null
/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/d1_mini -I/Users/delhoume/Documents/Arduino/libraries/FastLED /Users/delhoume/Documents/Arduino/libraries/FastLED/FastLED.cpp -o /dev/null
/Users/delhoume/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/Users/delhoume/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/k9/q90x0gwj19g_pll7tywc34_h0000gn/T/arduino_build_700693/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 -I/Users/delhoume/Library/Arduino15/packag [... cut ...]
/Users/delhoume/Documents/Arduino/libraries/FastLED/FastLED.h:464:25: note:   template argument deduction/substitution failed:
/Users/delhoume/Documents/Arduino/libraries/FastLED/FastLED.h:315:56: note:   cannot convert '& c' (type 'NEOPIXEL<3u>*') to type 'CRGB*'
   return addLeds(&c, data, nLedsOrOffset, nLedsIfOffset);
                                                        ^
Using library FastLED at version 3.2.7 in folder: /Users/delhoume/Documents/Arduino/libraries/FastLED 
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

NimmLor, 2019-06-07:
Hi, I'am getting similar errors too. The only working combination that works for me is fastled 3.2.6 in combination with esp8266 2.4.2 in Arduino 1.8.7. Any newer library/board version leads to tons of errors. 
However a friend of mine got it to run with esp8266 2.5.2 in Arduino 1.8.8.

focalintent, 2019-06-07:
Fixed in 3.2.8

closed 2019-06-07 by focalintent