FastLED/FastLED #813: 3.2.8 causes LED strip to be just white and uncontrollable on esp8266

NimmLor, 2019-06-07:
Hi, the new 3.2.8 solved the uploading issue on the esp8266 for be, but now my leds stay 100% brightness white and they can't be controlled. I've tested the DemoReel100 and some of my sketches and they all just turn the strip white. 
With 3.2.6 everything worked fine for me.

Tested with FastLED 3.2.8, esp8266 2.4.2 and 2.5.2, Arduino 1.8.7 and 1.8.9

InvisiBug, 2019-06-07:
I'm having this issue too. Just updated to 3.2.8 and using a node mcu

focalintent, 2019-06-08:
I’ll have to take a look over the weekend - I took a couple of pull requests for unrelated platforms but they may have broken something elsewhere.

hb020, 2019-06-08:
FWIW, 
Same symptoms as: https://forum.arduino.cc/index.php?topic=462791.0 (CPU clock problem)

Just started playing with this library, so cannot compare to an older working config, but I haven't yet found a combination that works:
FastLED 3.2.8, esp8266 2.5.1, Arduino 1.8.9: fail
FastLED 3.2.6, esp8266 2.5.0, 2.5.1, 2.5.2, Arduino 1.8.9: all fail.
Maybe the problem is in 2.5.x, or in any of the other libraries I/we use?
I personally use :

- ESP8266WiFi.h
- DNSServer.h
- WiFiClient.h
- ESP8266mDNS.h
- ESP8266HTTPUpdateServer.h
- WiFiManager.h

and I use 

#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_INTERRUPT_RETRY_COUNT 1
#include "FastLED.h" 

On the latest software (FastLED 3.2.8, esp8266 2.5.2), with a scope I have measured:
Signal period for WS2811_400 = 33 to 34us
Signal period for WS2812B = 16 to 17us

That is very far from the 1.25 us in the specs (for the WS2812B).

hb020, 2019-06-08:
Related to #782?

focalintent, 2019-06-08:
No this is something else - I took a pull request from someone who didn’t check their code well and it’s introduced problems - I have to get them fixed - hopefully at some point today.  I’m going to have to start being a lot stricter about what I’ll take.

mvoss96, 2019-06-08:
i can confirm that 3.2.8 does not work on esp32. All LEDS go full white. Downgrading to 3.2.6 fixed the problem

commit ca9e40b4 (2019-06-09): Fix #813 -- further fix for the C_NS macro added by the nrf52 changes
closed 2019-06-09 by focalintent