FastLED/FastLED #814: 3.2.8 breaks on Adafruit Trinket M0 (won't compile)

balbinus, 2019-06-08:
Hello,

It appears that a very basic Arduino sketch (the "FirstLight" example, no modifications) won't compile for the Trinket M0 using version 3.2.8, but did compile on 3.2.6. Replicated in both the Arduino editor and its Library manager, and using strict version requirements on PlatformIO.

It may be related to commit 403464a499 / #751, as it's the most recent change to this file? From my (very basic) understanding of ARM ASM, it seems like it expects 8 bit values but gets fed larger integers, like 654. I'm not really sure of this analysis, just trying to help ;) 

If you need any more information, please ask :)

```
/home/vincent/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/vincent/arduino-1.8.1/hardware -hardware /home/vincent/.arduino15/packages -hardware /home/vincent/sketchbook/hardware -tools /home/vincent/arduino-1.8.1/tools-builder -tools /home/vincent/arduino-1.8.1/hardware/tools/avr -tools /home/vincent/.arduino15/packages -built-in-libraries /home/vincent/arduino-1.8.1/libraries -libraries /home/vincent/sketchbook/libraries -fqbn=adafruit:samd:adafruit_trinket_m0 -vid-pid=0X239A_0X801E -ide-version=10801 -build-path /tmp/arduino_build_364769 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.openocd.path=/home/vincent/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.bossac.path=/home/vincent/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS-Atmel.path=/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -verbose /home/vincent/sketchbook/libraries/arduino_153452/examples/FirstLight/FirstLight.ino
/home/vincent/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/vincent/arduino-1.8.1/hardware -hardware /home/vincent/.arduino15/packages -hardware /home/vincent/sketchbook/hardware -tools /home/vincent/arduino-1.8.1/tools-builder -tools /home/vincent/arduino-1.8.1/hardware/tools/avr -tools /home/vincent/.arduino15/packages -built-in-libraries /home/vincent/arduino-1.8.1/libraries -libraries /home/vincent/sketchbook/libraries -fqbn=adafruit:samd:adafruit_trinket_m0 -vid-pid=0X239A_0X801E -ide-version=10801 -build-path /tmp/arduino_build_364769 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.openocd.path=/home/vincent/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino -prefs=runtime.tools.bossac.path=/home/vincent/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS-Atmel.path=/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -verbose /home/vincent/sketchbook/libraries/arduino_153452/examples/FirstLight/FirstLight.ino
Using board 'adafruit_trinket_m0' from platform in folder: /home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0
Using core 'arduino' from platform in folder: /home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0
Detecting libraries used...
"/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Trinket M0"'  "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/cores/arduino" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/variants/trinket_m0" "/tmp/arduino_build_364769/sketch/FirstLight.ino.cpp" -o "/dev/null"
"/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Trinket M0"'  "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/cores/arduino" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/variants/trinket_m0" "-I/home/vincent/sketchbook/libraries/arduino_153452" "/tmp/arduino_build_364769/sketch/FirstLight.ino.cpp" -o "/dev/null"
"/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Trinket M0"'  "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/cores/arduino" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/variants/trinket_m0" "-I/home/vincent/sketchbook/libraries/arduino_153452" "/home/vincent/sketchbook/libraries/arduino_153452/FastLED.cpp" -o "/dev/null"
"/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Trinket M0"'  "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/cores/arduino" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/variants/trinket_m0" "-I/home/vincent/sketchbook/libraries/arduino_153452" "/home/vincent/sketchbook/libraries/arduino_153452/bitswap.cpp" -o "/dev/null"
"/home/vincent/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Trinket M0"'  "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vincent/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/cores/arduino" "-I/home/vincent/.arduino15/packages/adafruit/hardware/samd/1.3.0/variants/trinket_m0" "-I/home/vincent/sketchbook/libraries/arduino_153452" "/home/vincent/sketchbook/libraries/arduino_153452/colorpalettes.cpp" -o "/dev/null"
[... cut ...]
/home/vincent/sketchbook/libraries/arduino_153452/platforms/arm/d21/../common/m0clockless.h:279:19: error: impossible constraint in 'asm'
       M0_ASM_ARGS);
                   ^
/home/vincent/sketchbook/libraries/arduino_153452/platforms/arm/d21/../common/m0clockless.h:314:8: error: impossible constraint in 'asm'
       );
        ^
Using library arduino_153452 at version 3.2.8 in folder: /home/vincent/sketchbook/libraries/arduino_153452 
exit status 1
Error compiling for board Adafruit Trinket M0.
```

focalintent, 2019-06-09:
It turns out this was caused by #318 - which was generating impossibly high values for the number of clocks involved.  I'm checking in a fix for that shortly and will spin 3.2.9 to take care of this.

closed 2019-06-09 by focalintent

balbinus, 2019-06-10:
Awesome! Thank you very much :) :+1: