FastLED/FastLED #818: FastLED not working with ESP8266 D1 Mini, WS2812B, and level shifter

shawheenattar, 2019-06-13:
I've been trying to get my ESP8266 board to interface with a WS2812B LED strip with FastLED, but I haven't been able to get it to work properly. After research I realized I may need a level shifter, so I got the SN74AHC125N to shift from 3.3V to 5V for the strip. This also did not work. I try to run the color palette example from the FastLED library but all the lights just become a bright white, with and without the level shifter. 

Then, I tried to use the NeoPixel library, and it worked without needing any level shifter. I also tried the FastLED library with my Arduino Mega and the same strip, and it worked perfectly. 

So, either the FastLED library isn't working with my ESP8266 board (I tried it with multiple boards that I own) or my level shifter isn't working. 

I've attached a picture of my wiring (bottom red and black cables are coming from 5V2A power supply, and left cables go to led strip). Does anything look wrong with it? Would really appreciate any help.

[image]

focalintent, 2019-06-13:
What version of FastLED do you have installed - there was a bug for a couple days that would send leds all white and should be fixed with version 4.2.9

shawheenattar, 2019-06-13:
Wow, just my luck. I started this project and updated my libraries a few days before 3.2.9 was released. Just updated it and its working now. Thanks man

closed 2019-06-13 by shawheenattar

sadler1981-Can, 2022-03-15:
I see this has been a few years, and I don't know what the pinout of your level shifter is, but typically you need to bring 3.3 volts to the level shifter from the ESP board, which from the picture I see that hasn't been done.