FastLED/FastLED #819: Making CRGB::operator bool() explicit would be useful

ssilverman, 2019-06-17:
It's easy to assign CRGB to an integral value for a number of reasons, however, if you aren't careful to use your own, for example, `uint32 toColor` function, the `bool()` operator is used.

My suggestion is to make `CRGB::operator bool()` `explicit`.

Ref: See mention of the Safe Bool idiom mentioned in the "Explicit Conversion Operators (C++11)" section at: https://stackoverflow.com/a/16615725.

kriegsman, 2019-06-17:
Thanks for the suggestion; I think it's a great idea.
I want to think about how and in what ways this is going to change the (compile-time or run-time) behavior of existing code, but overall I really like this idea.

ssilverman, 2019-06-17:
Yes, a large enough set of users means even testing things that "shouldn't" have an effect. :D

Having said that, I'm willing to bet adding `explicit` here will find usage errors in code that attempts to assign to any integral value, having an unintended consequence.

ssilverman, 2022-12-12:
One other note is that implicit conversions will need to be explicitly cast or used as a Boolean expression. See here: https://github.com/ssilverman/QNEthernet/blob/master/README.md#operator-bool-and-explicit

If that caveat is fine, then let’s get this in.

PR #1450 (2023-01-22): Make CRGB::operator bool() explicit
Resolves #819
closed 2023-01-22 by kriegsman