FastLED/FastLED #824: Can't declare more than 3 strips ?

Sylvian, 2019-06-23:
Hi there,

just an issue while trying to compile, I'm using :

`void setup() {
  FastLED.addLeds<WS2812, 9, GRB>(leds, 0, 3);
  FastLED.addLeds<WS2812, 10, GRB>(leds, 3, 3);
  FastLED.addLeds<WS2812, 11, GRB>(leds, 6, 3);
  FastLED.addLeds<WS2812, 12, GRB>(leds, 9, 3);
}`

and i'm having a compilation error :

`C:\Users\Dr S_\Documents\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h: In member function 'showPixels':

C:\Users\Dr S_\Documents\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:167:2: internal compiler error: Segmentation fault

  }

  ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Utilisation de la bibliothèque FastLED version 3.2.8 dans le dossier: C:\Users\Dr S_\Documents\Arduino\libraries\FastLED 
exit status 1
Erreur de compilation pour la carte Arduino Nano`

If I remove one line into the void setup { ... } it's just working fine, can someone tell me how to declare more than 3 strips without this issue ?

Thanks for your help.

focalintent, 2019-06-23:
You don’t mention what hardware platform you are compiling for.  And unfortunately it sounds like you are hitting a bug in gcc (which is the compiler) - which there may not be much I can do about. 

I do know that plenty of folks have declared more than three controllers without issue.

Sylvian, 2019-06-23:
Hi again,

Thabk you for answering me so fast :)
I’m using an arduino nano.

If i didn’t find a way by myself maybe I can try to paste the entire code here ?

Sylvian, 2019-06-24:
Here is the entire code : 

> #include <FastLED.h>
> CRGB leds[12];
> void setup() {
>   FastLED.addLeds<WS2812, 9, GRB>(leds, 0, 3);
>   FastLED.addLeds<WS2812, 10, GRB>(leds, 3, 3);
>   FastLED.addLeds<WS2812, 11, GRB>(leds, 6, 3);
>   FastLED.addLeds<WS2812, 12, GRB>(leds, 9, 3);
> }
> void loop() {
> 
> }

As soon as I remove one of the four lines into the setup it works, I can't explain it :S

focalintent, 2019-06-24:
It is a bug with the compiler - you have to follow the instructions in the link it gives to file a bug with the compiler - not with FastLED.

closed 2019-08-12 by focalintent