FastLED/FastLED #827: ESP32 Color Order

jordanadania, 2019-06-25:
`FastLED.addLeds<WS2812B, 23, GRB>(leds, NUM_LEDS);
FastLED.addLeds<WS2811, 32, RGB>(leds, NUM_LEDS);`

With that code, the WS2812B strip runs correctly, but the WS2811 code swaps the Red and Green. Switching the WS2811 to GRB has no effect.

Putting the WS2811 line before the WS2812B gives opposite results. The WS2811 strip will run correctly, but the WS2812B strip would have Red and Green swapped. Switching the WS2812B to RGB has no effect.

focalintent, 2019-06-26:
@samguyer I can see the problem - it's this line:

https://github.com/FastLED/FastLED/blob/master/platforms/esp/32/clockless_i2s_esp32.h#L628

```
            ClocklessController * pController = static_cast<ClocklessController*>(gControllers[i]);
```

Because you are keeping a single array of clockless controller objects and you're doing a forcible static cast here, it's overriding the RGB_ORDER template parameter - and because CPixelController is _not_ virtual (for optimization reasons) - it's basically throwing away the different RGB ordering for different sets of pixels when using the i2s output.

You're going to need to do some juggling w/virtual methods to make sure that you're getting the correct color ordering for different strips. 

Similarly on the rmt side:

https://github.com/FastLED/FastLED/blob/master/platforms/esp/32/clockless_rmt_esp32.h#L484

                    // -- Refill the half of the buffer that we just finished,
                    //    allowing the other half to proceed.
                    ClocklessController * controller = static_cast<ClocklessController*>(gOnChannel[channel]);
                    controller->fillHalfRMTBuffer();

that static cast is basically throwing away the RGB_ORDER template information.

samguyer, 2019-06-26:
I think it's ok in the RMT controller because the fill method is virtual.
The I2S controller already has the limitation that all strips must be
identical.

jordanadania, 2019-06-26:
I'm more on the noob side of things, so if the answer has been posted in your language, just let me know in my language so I can begin to translate.

focalintent, 2019-06-26:
The answer is that there’s a bug in FastLED that needs to get fixed :) just a question of time

samguyer, 2019-06-26:
@focalintent 
Huh. It looks like I did not make those methods virtual, which probably explains why it doesn't work.
I believe the solution is to make fillHalfRMTBuffer and startOnChannel virtual methods. That was my intent, but somehow it didn't make it in there.

samguyer, 2019-06-26:
@focalintent You'll appreciate this: I've got two laptops, side by side, one debugging FastLED and the other debugging saf. :-)

jordanadania, 2019-06-26:
I found a real life bug that isn't just me forgetting something small?!

This is going on my resume!

jordanadania, 2020-05-26:
Apparently, it's not fixed??

jordanadania, 2020-05-26:
I thought it was fixed, but you would have to ask old me that. I personally dont trust him very much.

jordanadania, 2020-05-26:
Also, my first 2 pixels on every strip are wrong color order as well.

jordanadania, 2020-05-26:
I wouldn't trust old me; it's definitely not working right now.

Also, my first 2 pixels on every strip are wrong color order; swapped with RGB and GRB cause blue is correct for everything.

https://github.com/jordanadania/fastled-newFlow

This is the code; it's small. The setup is the only thing relevant anyway.

jordanadania, 2020-05-26:
Do you want me to open a brand new ticket?

samguyer, 2020-05-26:
Nah, let's just keep this ticket open. When I looked at the code I thought to myself "How could this possibly be fixed?!?!" :-)

samguyer, 2020-05-27:
@kriegsman Do you have any thoughts on how to handle this situation? I can explain the problem in more detail if you want.

jordanadania, 2020-05-27:
on the 8266, i can set 3 of them to parallel output, and then assign the 4th normally and it works. maybe see how its done on 8266? they are both parallel output i feel like.

samguyer, 2020-05-27:
@jordanadania Yeah, it's a totally different method of sending the bits. The short version is that the ESP32 driver is called by a periodic interrupt, so we need to pass the color order information through that interrupt interface, which is hard because it's a template parameter, not a runtime value.

jordanadania, 2020-06-04:
> I think it's ok in the RMT controller because the fill method is virtual. The I2S controller already has the limitation that all strips must be identical.

How do I switch to the RMT controller?

BradMcc522, 2020-08-11:
I think I ran into a similar issue.  I have multiple strips, all WS2815.  3 of them are GRB and work perfect.  The 2 strips that are RGB and despite being set as RGB they have the red and green switched.

jordanadania, 2020-08-11:
This is due to clockless. Its why i use nothing but 8266sSent from my T-Mobile 4G LTE Device

agrath, 2021-05-06:
I have been watching this issue for a while as I have a project that has two strips connected, one using BGR and another RGB on an ESP32 (arduino)
The second strip was not working well for our project so we disabled it as it was for debugging mostly anyway.
We additionally observed that the first two pixels on the second strip were also the wrong colour.

I have just retested using fastled 3.4 and can't reproduce the issue anymore, both strips are working with their correct colour spaces.

jordanadania, 2021-05-06:
Maybe it is time to switch back to esp32!

I will close this until I can recreate the problem.

closed 2021-05-06 by jordanadania

talkncloud, 2021-09-05:
tested today, esp32 on 3.4, not fixed, it is most definitely GRB as someone else commented on.

jordanadania, 2021-09-05:
Thanks for saving me the trouble.Sent from my T-Mobile 4G LTE Device