FastLED/FastLED #837: ESP2866ex and WS2812B Invalid Pin specified error

dr-ace07, 2019-07-12:
In file included from C:\Users\Dr. Ace\Desktop\Round-LED-Clock-master\Round-LED-Clock-master\Round-LED-Clock\Round-LED-Clock.ino:19:0:

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.010

 #    pragma message "FastLED version 3.002.010"

                     ^

In file included from C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Dr. Ace\Desktop\Round-LED-Clock-master\Round-LED-Clock-master\Round-LED-Clock\Round-LED-Clock.ino:19:

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/fastspi.h:115:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/FastLED.h:48:0,

                 from C:\Users\Dr. Ace\Desktop\Round-LED-Clock-master\Round-LED-Clock-master\Round-LED-Clock\Round-LED-Clock.ino:19:

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<6u>':

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_esp8266.h:21:49:   required from 'class ClocklessController<6, 20, 50, 30, (EOrder)66u, 0, false, 50>'

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/chipsets.h:571:7:   required from 'class WS2812Controller800Khz<6u, (EOrder)66u>'

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/FastLED.h:105:52:   required from 'class WS2812B<6u, (EOrder)66u>'

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 6u; EOrder RGB_ORDER = (EOrder)66u]'

C:\Users\Dr. Ace\Desktop\Round-LED-Clock-master\Round-LED-Clock-master\Round-LED-Clock\Round-LED-Clock.ino:82:57:   required from here

C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

Multiple libraries were found for "FastLED.h"
 Used: C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED
 Not used: C:\Users\Dr. Ace\Documents\Arduino\libraries\FastLED-master
exit status 1
Error compiling for board Generic ESP8266 Module.

**CODE:**

/*
  WiFi connected round LED Clock. It gets NTP time from the internet and translates to a 60 RGB WS2812B LED strip.

  If you have another orientation where the wire comes out then change the methods getLEDHour and getLEDMinuteOrSecond

  Happy programming, Leon van den Beukel, march 2019

  ---  
  NTP and summer time code based on:
  https://tttapa.github.io/ESP8266/Chap15%20-%20NTP.html 
  https://github.com/SensorsIot/NTPtimeESP/blob/master/NTPtimeESP.cpp (for US summer time support check this link)  
  
*/

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiUdp.h>
#include <FastLED.h>
#define DEBUG_ON

const char ssid[] = "Matrix";        // Your network SSID name here
const char pass[] = "dec1271995";        // Your network password here
[... 284 lines cut ...]
  Serial.print(" day of week: ");
  Serial.print(currentDateTime.dayofweek);
  Serial.print(" summer time: ");
  Serial.print(summerTime());
  Serial.print(" night time: ");
  Serial.print(night());  
  Serial.println();
#endif  
}

boolean summerTime() {

  if (currentDateTime.month < 3 || currentDateTime.month > 10) return false;  // No summer time in Jan, Feb, Nov, Dec
  if (currentDateTime.month > 3 && currentDateTime.month < 10) return true;   // Summer time in Apr, May, Jun, Jul, Aug, Sep
  if (currentDateTime.month == 3 && (currentDateTime.hour + 24 * currentDateTime.day) >= (3 +  24 * (31 - (5 * currentDateTime.year / 4 + 4) % 7)) || currentDateTime.month == 10 && (currentDateTime.hour + 24 * currentDateTime.day) < (3 +  24 * (31 - (5 * currentDateTime.year / 4 + 1) % 7)))
  return true;
    else
  return false;
}

boolean night() {
  
  if (currentDateTime.hour >= NIGHTCUTOFF && currentDateTime.hour <= MORNINGCUTOFF) 
    return true;    
}

dr-ace07, 2019-07-12:
@focalintent can you please take a look for me. :(

focalintent, 2019-07-12:
It sounds like the board is using the raw pin ordering / which means you need to figure out what the actual gpio pin number is - and with the raw ordering - gpio 6 is not a valid pin (there’s two numberings for pins - there’s the raw low level gpio number for a pin - but then board makers sometimes put their own numbers and say “pin 5 refers to gpio 3” - I don’t have a fully solid set of these mappings for all the board providers.  You can look at the file platforms/esp/8266/fastpin_esp8266.h to see the set of defined pins for the raw mapping.

closed 2019-08-12 by focalintent