FastLED/FastLED #840: SparkFun Pro nRF52840 Mini + Neopixels locking up when calling FastLED.show() too quickly with more than 32 LEDs

masonevans, 2019-07-17:
I have a new SparkFun Pro nRF52840 MIni and I'm trying to get it up and running with Neopixels + FastLED.  I've got it most of the way there but I'm running into a small issue with things locking up when I call FastLED.show() too quickly if I have more than 32 LEDs set up.  

To start, I'm using the Adafruit nRF52 board definition (v0.10.1 because FastLED won't compile against the newest, v0.11.1, probably a separate issue).   I'm using the board definition and variant available here:  https://github.com/sparkfun/nRF52840_Breakout_MDBT50Q/tree/master/Firmware/Arduino

I made one change to the board definition, they define it as an NRF52840_FEATHER but I don't think that's correct because the pinouts are different, so I changed the board definition to NRF52840_PCA10056 as that matches the pinouts on this board.  I am running the data line off a port which support high speed GPIO.

Everything seems to work ok if I'm trying to drive up to 32 pixels.  But when I get beyond that, the board locks up relatively quickly if I call FastLED.show() without a rate limit.   When it locks up, the LEDs stop refreshing and it stops writing to the Serial console (although Bluetooth DOES continue to work, I can connect and send commands).  If I wrap FastLED.show() in an EVERY_N_MILLISECONDS(3) call then it seems to run fine forever. 

I've tried adjusting the delay how quickly show() is called.  With 32 LEDs I don't need to make the call at all.  With 33 LEDs I need to set the delay to at least 3ms to keep it from locking up (1 and 2 don't work).  3ms also seems to work up to at least 64 LEDs, so from that perspective the delay and number of LEDs beyond 32 don't seem to be related.

henrygab, 2019-07-19:
@masonevans Well written bug report.  This sounds like it will either be devilishly hard, or relatively simple, to track down.  I will not be able to do a deep investigation until mid-August.  However, your data points do suggest there is a mutual-exclusion issue.  I'll see what a code review turns up...

Two questions:
1. Does the lockup occur even if the LEDs are entirely disconnected?  (no power, no board connection)
2. Does this occur on many pins?  (e.g., not only pin 5)

For my own reference: (( NEOPIXEL == WS2812 == clockless ))

masonevans, 2019-07-22:
> Does the lockup occur even if the LEDs are entirely disconnected? (no power, no board connection)

I just tried it, I disconnected everything from the board except for a USB cable so I could monitor the Serial output, and I see the same behavior.  64 pixels will lock up immediately, fewer will take longer to lock up, but the Serial output does eventually stop.

> Does this occur on many pins? (e.g., not only pin 5)

Yes, I just tried  a few others pins, even with the pixels disconnected, and they all  exhibited the same behavior.

henrygab, 2019-07-25:
Ok,  that's very helpful:
* hardware not required to repro (so not likely to be power related)
* large LED arrays increase likelihood
* ultra-simple repro ino provided

It's probably an error in the mutual exclusion class, or I forgot a wait somewhere.  I am still without my equipment for another week and a half,  but this is top of mind.

henrygab, 2019-08-01:
Importantly, the current clockless implementation does **_not_** currently wait for the PWM hardware to be done sending the bits.  Rather, it uses PWM hardware and an ISR with callback to indicate completion.

<details>
  <summary>I'm going to use this post to store my ongoing notes, since I'm without my normal equipment for a while longer....</summary>

  <p>

* LED Hardware is not required to repro, so unlikely to be power-related.
* Larger LED arrays increase likelihood of repro
* SparkFun Pro's Arduino port 5 corresponds to nRF52 port 0.5
* Clockless on nRF52 uses the PWM DMA-based hardware

There are three important concepts in use:
1. Exclusive access to the sequence buffer used by PWM hardware:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L49
2. Exclusive access to use PWM hardware:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/arbiter_nrf52.h#L37-L38
3. Ensuring that there's sufficient pause to cause clockless LEDs to latch the data: https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L50

</details>

<details>
  <summary>Sequence buffer exclusive access</summary>

  <p>

`showPixels()` immediately infinite-loops until `s_SequenceBufferInUse` is zero:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L162-L164

It sets this value to one in a cache-safe manner, on that same thread:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L241-L246

And sets the value to zero in the ISR, when the PWM hardware stops:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L134-L138

Based on the above, it does not appear that the sequence buffer could ever be used by more than a single in-progress showPixels() call.  However, the ISR does (purposely) set this to zero prior to releasing the PWM class object, so there _is_ a timing window when the sequence buffer is not in use, but the PWM arbiter is still marked in use.

</details>

<details>
  <summary>TODO: Does mWait properly ensuring wait time?</summary>

  <p>

TODO...

</details>

<details>
  <summary>TODO: Does PWM arbiter class have timing window?</summary>

  <p>

TODO...

</details>

henrygab, 2019-08-02:
Update:
1. I have locally reproduced the issue, using adafruit's nrf52840 feather express.
2. If set to 40 LEDs, the repro locks up after ~ 0x2000 iterations of `showLed()`.
3. If set to 80 LEDs, the code fails substantially immediately, leaving the NEOPIXEL a solid green... which is really odd and suggests more than one issue.  (I've not put a scope on this yet.)
4. If I fix the _(unrelated?)_ `CMinWait<>` issue, by calling `.mark()` in the ISR, then the 40 LED strand continues to run overnight.  However, this **_may_** just be hiding the issue rather than fixing it.

Next steps:
1. Get scope trace from 80-LED strand
2. Add infinite loops where runtime assertions are noted as comments in code
3. Use J-Link to see state of locked-up system

If the above does not resolve the issue, then I can change the code to be synchronous, and not return from `showLeds()` until the PWM peripheral is finished.  This is no worse than most chipsets, but I would prefer to avoid this....

masonevans, 2019-08-02:
@henrygab  - I can confirm that I have also see the "green first pixel" issue.  I have noticed that doing something like immediately filling the strip in my setup() function will cause the first pixel to be colored green.  From what I can tell, the next time I do anything with the strip the first pixel gets updated to the correct color and that issue doesn't come up again.

henrygab, 2019-08-05:
@masonevans Potentially very good news....

I have found potential cause(s) and have a private branch that appears (_at least preliminarily_) to fix this problem.

<details>
  <summary>The long road...</summary>

  <p>

After the [manual reviews](https://github.com/FastLED/FastLED/issues/840#issuecomment-517340370) listed earlier didn't shed light on the problem, and because some of my equipment is still not setup, I needed more data from a solid repro.

As I have a J-Link, I used Segger's SysView, which AdaFruit appears to have provided built-in support for on the nRF52.  So I added custom events in SysView.

This was not as simple as it should have been.  Pain points include:
* Documentation on how to add a module is great, in the same way a jumble of jigsaw puzzle pieces is a picture.  Each piece exists, but how they fit together is unclear.
* Documentation presumes use of Segger Embedded Studio, which doesn't translate well to Arduino.
* Custom events … The SystemView tool itself wants files to be written to the app installation directory, which is not writable on Windows.  Luckily, there's a way to provide the format strings programmatically when registering a module.
* The SYSVIEW API limits string length to 128 bytes (by default).  This is not normally a real issue.   Except that this limit is also applied to strings used when registering a module … which is the way that format strings are registered for the module.  Therefore, ***ALL** module information and format strings must (combined) be under 128 bytes.
* If too many events defined, the output wouldn't be formatted properly (silently failing).
* If defined events fired too often, events would be lost.

Eventually I got lucky: a build happened to have just the right events and a minimal repro, suggesting what close to the "last thing" was that happened, just before the hard-lock....

</details>

<details>
  <summary>When does the hard-lock occur?</summary>

  <p>

The following sequence of events caused a repro:
1. showPixels() is called first time, starts the PWM peripheral on the task, and returns
2. showPixels() is called a second time, and stalls because the PWM peripheral is still sending bits across the wire
3. the PWM peripheral finishes and thus the interrupt fires, and the ISR frees the PWM arbiter and sequence buffers for re-use.
4. When the ISR exits, it goes back to the loop (the second showPixels() call) … and then `nothing happens`.

</details>

<details>
  <summary>Potential cause?</summary>

  <p>

I really wish it was some complex, hardware-based bug.  It now seems much more likely a combination of a few things.

The following may be the minimum related items:
1. I failed to mark the s_SequenceBufferInUse variable as volatile:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L49
I knew it should be volatile.  This is a classic example of when volatile is necessary:
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L162-L164

2. I failed to call mWait.mark() in the ISR handler (potentially failing to allow latching of data before another update is sent).
https://github.com/FastLED/FastLED/blob/13a9e0cc0999025b7ffdf92932bc6efb2199f8ea/platforms/arm/nrf52/clockless_arm_nrf52.h#L134

3. The order of operations in the ISR may need tweaking.

4. changes to s_SequenceBufferInUse may need memory barrier semantics.  I can't be sure, but I'm going to add full barrier semantics here anyways.

</details>

Anyways, I have a branch that has run for over an hour with 140 LED strands (none actually attached, but it's still running....).  I am looking to reduce / minimize the code changes needed, before generating a PR.

Let me know if you want early access and can test from either / both the dirty dev branch, or the cleaner PR branch before I submit.

henrygab, 2019-08-05:
https://github.com/henrygab/FastLED/tree/MinFix840

@masonevans I think this is the minimal fix needed.
I've constrained the changes to a single file … [clockless_arm_nrf52.h](https://raw.githubusercontent.com/henrygab/FastLED/MinFix840/platforms/arm/nrf52/clockless_arm_nrf52.h).

Are you willing to give it a go to verify it resolves this issue for you?

masonevans, 2019-08-05:
@henrygab Just tested this out and it seems promising so far.  I let it run for a bit with 64 LEDs configured but without any LEDs attached and that ran fine.  I'm now letting it run with a strip plugged in with 60 LEDs (all I have handy at the moment) and it's been running for about 15 minutes with no lock ups.  

I'll let it run for a few hours and leave a comment if it locks up.

henrygab, 2019-08-05:
@masonevans I'm thrilled to hear that -- awesome!

Can you please give positive confirmation if your setup is still running after at least eight hours? I'm also running this set to 140 LEDs (none actually connected).  Since both previously reproduced the issue essentially immediately, having those two data points in combination with the SysView trace will provide extremely high confidence in the fix.

masonevans, 2019-08-06:
@henrygab  Running for 10 hours with no issues now

henrygab, 2019-08-06:
and nearly 16 hours here,  I think it's about time for a PR.

@masonevans : Thank you for your patience and clear/minimal repro.  👍

masonevans, 2019-08-06:
@henrygab  I'm at about 20 hours now as well with no problems.  Thanks for having a look at this!

henrygab, 2019-08-10:
@masonevans -- Would be willing to validate a potential fix for #856 (first LED wrong color first time only)?  See
https://github.com/henrygab/FastLED/commit/ded19a59cca36a05dcf93cd2095ce4a6f68e15e2.

The potential fix is to add the following four lines (two of which are comments) to file [platforms/arm/nrf52/clockless_arm_nrf52.h](https://github.com/henrygab/FastLED/blob/ded19a59cca36a05dcf93cd2095ce4a6f68e15e2/platforms/arm/nrf52/clockless_arm_nrf52.h#L159-L162) at function init():
```C++
    // to avoid pin initialization from causing first LED to have invalid color,
    // call mWait.mark() to ensure data latches before color data gets sent.
    startPwmPlayback_InitializePinState();
    mWait.mark();
```
While I couldn't get my scope connected to pin 8 to prove the cause,
it appears to prevent any the first LED color issue, without affecting
later calls to `showPixels()`.

Can you please let me know, whether you don't test it, or the results you get when testing it?

Thanks!

PR #857 (2019-08-13): Fix nRF52 race condition / hard lock
Fixes #840.
<details>
    <summary>A textbook case of where `volatile` is necessary</summary>

<p>

The variable `s_SequenceBufferInUse` is modified in an ISR.
However, the variable was **_not_** marked `volatile`:
```C++
virtual void showPixels(PixelController<_RGB_ORDER> & pixels) {
    while (s_SequenceBufferInUse != 0);
```

Because it wasn't marked `volatile`, the compiler was allowed to use the first value it read.  Using [compiler explorer](https://gcc.godbolt.org/) set to `ARM gcc 7.2.1 (none)` with optimizations (e.g., `-O5`), you would get somethi [... cut]
closed 2019-08-13 by focalintent