FastLED/FastLED #841: beatsin16 sometimes crashing on ESP8266

CodilX, 2019-07-18:
I have an array of 6 effects that each play for a minute and then switch to another one. Occasionally my ESP8266 (NodeMCU v3) would crash, sometimes after a few minutes, other times after half an hour or more. I checked the error output via serial.

```
Exception (0):
epc1=0x4000e25d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffd20 end: 3fffffd0 offset: 01a0

stack
3ffffec0:  00000000 00000000 4bc6a7f0 00000000  
3ffffed0:  3ffee928 0000001c 401066d0 001573d4  
3ffffee0:  00000000 0000032f 00000000 00000000  
3ffffef0:  18cded2f 0000001c 3ffee87c 40203164  
3fffff00:  3ffee8d4 0000001c 0000001c 00000063  
3fffff10:  3ffee93c 3ffee93c 3ffee92c 402031a5  
3fffff20:  00000032 3ffee93c 00000000 40203280  
3fffff30:  002f2232 3ffee8d4 3ffee904 40202836  
3fffff40:  40203100 3ffee8d4 0000001c 00000001  
3fffff50:  00000000 00000001 3ffee93c 3ffee8b0  
3fffff60:  3ffee8a8 00000000 3ffee86c 3ffee8b0  
3fffff70:  3ffee8a8 3ffee86c 3ffee93c 40202ef4  
3fffff80:  fec8783c feefeffe feefeffe feefeffe  
3fffff90:  0000ea60 feefeffe feefeffe feefeffe  
3fffffa0:  00000000 00000000 00000001 3ffee98c  
3fffffb0:  3fffdad0 00000000 3ffee984 4020390c  
3fffffc0:  feefeffe feefeffe 3ffe860c 40100739  
stack

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```
I then decoded it and here's the following

```
Decoding 9 results
0x401066d0: millis at C:\Users\myUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_wiring.c line 183
0x40203164: CPixelLEDController(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at C:\Users\myUser\Documents\Arduino\libraries\FastLED/controller.h line 171
0x402031a5: CFastLED::countFPS(int) at C:\Users\myUser\Documents\Arduino\libraries\FastLED/FastLED.cpp line 266
0x40203280: CFastLED::show(unsigned char) at C:\Users\myUser\Documents\Arduino\libraries\FastLED/FastLED.cpp line 266
0x40202836: beat88 at C:\Users\myUser\Documents\Arduino\libraries\FastLED/controller.h line 171
:  (inlined by) beat16 at C:\Users\myUser\Documents\Arduino\libraries\FastLED/lib8tion.h line 920
:  (inlined by) beatsin16 at C:\Users\myUser\Documents\Arduino\libraries\FastLED/lib8tion.h line 951
:  (inlined by) juggle() at C:\Users\myUser\Documents\Arduino\TestLEDRing/test2.ino.ino line 247
0x40203100: CPixelLEDController(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at C:\Users\myUser\Documents\Arduino\libraries\FastLED/controller.h line 171
0x40202ef4: loop at C:\Users\myUser\Documents\Arduino\libraries\FastLED/controller.h line 171
0x4020390c: loop_wrapper at C:\Users\myUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x40100739: cont_wrapper at C:\Users\myUser\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81
```

Here's the function in my code where beatsin16 failed

```
void juggle() {
  fadeToBlackBy( leds, NUM_LEDS, 20);
  byte dothue = 0;
  for( int i = 0; i < 8; i++) {
    leds[beatsin16(i+7,0,NUM_LEDS)] |= CHSV(dothue, 200, 255); // <- this line failed (line 247)
    dothue += 32;
  }
}
```

I proceeded to update FastLED to version 3.2.10 and the same exact problem occurred and this error log is from the latest version.

tobi01001, 2019-07-18:
Looks like you sine wave is going from 0 to NUM_LEDs (being capped at NUM_LEDS) while you actually want it to go from 0 to NUM_LEDS-1?

So depending on the calculation result you may just write out of bounds where the result is not predictable.

`leds[beatsin16(i+7,0,NUM_LEDS-1)]` should work.

CodilX, 2019-07-18:
> 
> 
> Looks like you sine wave is going from 0 to NUM_LEDs (being capped at NUM_LEDS) while you actually want it to go from 0 to NUM_LEDS-1?
> 
> So depending on the calculation result you may just write out of bounds where the result is not predictable.
> 
> `leds[beatsin16(i+7,0,NUM_LEDS-1)]` should work.

I'm dumb.. Thank you for spotting this error! Case closed :)

closed 2019-07-18 by CodilX