FastLED/FastLED #842: FastLED no longer compiles against latest Adafruit nRF52 board definitions (gcc 7-2017q4) due to invalid C++ code in FastLED

masonevans, 2019-07-18:
In v0.11.0, Adafruit switched the version of gcc that they are using for the nRF52 boards

See this commit: https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/bbe64fb9bb2ee2d8fbe6a7a11b8ae817379b7938#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9

Along with their package.json (https://adafruit.github.io/arduino-board-index/package_adafruit_index.json), comparing lines 3264 and 3590.

The effect of this change is to move nRF52 build toolchain from gcc 5.2 to gcc 7.

One of the changes between these two versions of gcc was to fix a gcc bug that was allowing reinterpret_casts to happen inside constexprs (according to the spec this is not valid).  The compiler change is now causing the following compilation error when compiling for any of the nRF52 boards (nRF52832 & nRF52840):

```
In file included from /Users/user/Library/Arduino15/packages/adafruit/hardware/nrf52/0.11.1/cores/nRF5/nordic/nrfx/mdk/nrf.h:93:0,
                 from /Users/user/Library/Arduino15/packages/adafruit/hardware/nrf52/0.11.1/cores/nRF5/WVariant.h:23,
                 from /Users/user/Library/Arduino15/packages/adafruit/hardware/nrf52/0.11.1/cores/nRF5/Arduino.h:40,
                 from /var/folders/c2/wdh0gcts5qn6rf1npbsnxf180000gn/T/arduino_build_147358/sketch/sketch_jul17a.ino.cpp:1:
/Users/user/Arduino/libraries/FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h: In static member function 'static constexpr NRF_GPIO_Type* __generated_struct_NRF_P0::r()':
/Users/user/Library/Arduino15/packages/adafruit/hardware/nrf52/0.11.1/cores/nRF5/nordic/nrfx/mdk/nrf52840.h:2826:38: error: reinterpret_cast from integer to pointer
 #define NRF_P0                      ((NRF_GPIO_Type*)          NRF_P0_BASE)
                                     ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/user/Arduino/libraries/FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:77:16: note: in expansion of macro 'NRF_P0'
         return NRF_P0;
                ^~~~~~
/Users/user/Arduino/libraries/FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h: In static member function 'static constexpr NRF_GPIO_Type* __generated_struct_NRF_P1::r()':
/Users/user/Library/Arduino15/packages/adafruit/hardware/nrf52/0.11.1/cores/nRF5/nordic/nrfx/mdk/nrf52840.h:2827:38: error: reinterpret_cast from integer to pointer
 #define NRF_P1                      ((NRF_GPIO_Type*)          NRF_P1_BASE)
                                     ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/user/Arduino/libraries/FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:82:16: note: in expansion of macro 'NRF_P1'
         return NRF_P1;
                ^~~~~~
```

The problematic code is here in fastpin_arm_nrf52.h:

```
struct __generated_struct_NRF_P0 {
      FASTLED_NRF52_INLINE_ATTRIBUTE constexpr static NRF_GPIO_Type * r() {
          return NRF_P0;
      }
  };
```

NRF_P0 is a macro which expands to 

```
#define NRF_P0                      ((NRF_GPIO_Type*)          NRF_P0_BASE)
```

and NRF_P0_BASE is just the memory address of P0

```
#define NRF_P0_BASE                 0x50000000UL
```

Unfortunately my C++ template metaprogramming skills are not good enough to try to fix this and submit a PR.

masonevans, 2019-07-18:
@henrygab possibly you have an idea here as you wrote the nRF52 port of FastLED?

henrygab, 2019-07-19:
Yes, I actually just noticed this, and have a patch I'm testing right now.
It seems that GCC became more standards-compliant (or pendantic, depending on your viewpoint) as to what is allowed in `constexpr`.  Specifically, they disallow constant integers being cast to pointers of any type, requiring the values to be cast at time-of-use.

I just finished running my final Travis-CI build, and have submitted a PR.

henrygab, 2019-07-19:
@masonevans , thanks for opening this issue.

You mentioned both nRF52832 and nRF52840... Out of curiosity, what boards are you using this on?

Would you be able, for your nrf52832 board, to verify each pin works as expected? (see `fastpin_arm_nrf52_variants.h`, where I enabled many boards, but most are untested)   It could be as simple as iteratively connecting some clockless LEDs to each pin, running the RGBCalibrate sample against the pin, and noting any pins that don't appear to work or don't seem to be accessible..

The boards I have:
* Adafruit Bluefruit nRF52840 Feather Express
* Nordic nRF52840 DK (aka PCA10056)

Being able to remove the 'untested' warnings from some of the other boards would be great.

masonevans, 2019-07-19:
@henrygab  sorry, I don't actually have an nRF52832, I just tried compiling against the board so I had some additional data points to put in the issue.  I think I tried that and started writing the issue before I realized that it was caused by the change in gcc version which would affect all the boards.

I'm actually running this on a SparkFun Pro nRF52840 Mini (https://www.sparkfun.com/products/15025), which is setup in my boards.txt as a PCA10056.  According to SparkFun the pinout should match between the two boards.   But If I get a minute I'll create a PR to add the board's pin definitions it to `fastpin_arm_nrf52_variants.h`.

I did have [one issue](https://github.com/FastLED/FastLED/issues/840) with the SparkFun board when running on Adafruit v0.10.1.  I was going to re-test after your PR and see if that issue persists with v0.11.1 of the Adafruit board definition.

PR #845 (2019-07-19): Fix #842 by increasing compliance with constexpr restrictions
Fix #842

Specifically, cannot cast integer values to pointers in a constexpr,
requiring use of intptr_t in the constexpr.  Callers of the constexpr
may then cast it at the non-constexpr location of use.

See https://stackoverflow.com/questions/10369606/constexpr-pointer-value
closed 2019-07-19 by focalintent