FastLED/FastLED #843: NS107S support

laforestlabs, 2019-07-18:
It look like new star is making a [new LED they are calling NS107S](https://cdn.hackaday.io/files/21293910999712/NS107S%20RGB%20LED%20Chip.pdf). This is an interesting option because the PWM rate is very high at 26 khz, making it useful for POV applications. Also it look like it comes in small size packages like 3535 and 2020. 

Is there any chance FastLED will support this LED?

focalintent, 2019-07-18:
The protocol looks identical to the APA102 - so nothing will probably need to be done to support it.

(Given the failure of the APA102 and SK clones to knock the clockless WS281x style chipsets off the throne, I suspect this will stay a niche led player, though)

focalintent, 2019-07-18:
Ah - it's just the APA107 which is the successor to the APA102. I wonder if they've fixed the signal degradation issue with the APA102 that requires lowering the clock speed the longer your strip of leds is...

dimitre, 2019-11-22:
Yes you can use the same protocol and I think they fixed the problem described here by Paul
https://www.pjrc.com/why-apa102-leds-have-trouble-at-24-mhz/