FastLED/FastLED #848: Enable TravisCI builds

henrygab, 2019-07-25:
See PR "Enable TravisCI builds #846".

Enabling TravisCI builds will help prevent build breaks (much like my own earlier failure).

The PR builds all platforms except for Teensy boards, and ensures all supported examples build for those platforms.

henrygab, 2019-08-01:
Is there any interest in this?

focalintent, 2019-08-01:
There is but it is a large enough set of changes that I don’t want to pull it in until I can spend some time with it here.

henrygab, 2019-08-03:
OK, thanks for the update.  Let me know if any questions.

henrygab, 2020-01-11:
With the addition of GitHub actions, and the continued maintenance requirements, I am going to close this for now.

henrygab, 2020-03-08:
Re-opening this, as it seems that a functioning CI check would provide immediate value, both for PRs in the official FastLED repository, as well as in any forks that enable TravisCI.

NOTE: Accepting this PR (or pulling once it is accepting) will *NOT* enable TravisCI builds by itself.

closed 2022-01-25 by henrygab