FastLED/FastLED #849: ESP8266 with WS2811 --> Flickering

tbnobody, 2019-07-26:
Hello,

after a longer time I upgraded the FastLED library within my project. It consists of several external libraries including:
* FastLED
* ESPAsyncWebserver
* ESP8266
* 100x WS2811 LEDs

When using the latest version of FastLED (3.2.10) I figured out, that my LEDs are flickering and generating artefacts. 
I went back every version until I found a working one. The problem occours with the change from 3.1.8 to 3.2.0. Especially this commit produced the error: https://github.com/FastLED/FastLED/commit/dcbf39933f51a2a0e4dfa0a2b3af4f50040df5c9#diff-c91a14657b986d84f0e5b32bca731424

I changed the WAIT_TIME back to 5 and everything works like a charm.
Due to the fact that this high WAIT_TIME is need by other people (but produces errors at me), is there a way to overwrite this value without changing the source code of the FastLED library? Maybe it's worth to add a define? (Currently I don't know any other way to overwrite this value from my source code)

focalintent, 2019-07-27:
There isn't an easy way with the current version of FastLED - the overhaul to the clockless controllers that I'm working on will allow that at some point in the future.

marmotton, 2019-11-23:
I'm not sure this is related, but I had some flickering issues with a WS2812B strip and a ESP8266, FastLED v3.3.2. I fixed it by setting `FastLED.setMaxRefreshRate( 100 )` after `FastLED.addLeds`. I guess that the value I chose (100Hz) depends on the number of LEDs in the strip.

stnkl, 2021-07-17:
> There isn't an easy way with the current version of FastLED - the overhaul to the clockless controllers that I'm working on will allow that at some point in the future.

Is there any update on this? It appears to me that using WS2812Bs with the ESP8266 is still broken in the latest release of FastLED manager. I am surprised nobody else is mentioning it anywhere. I was able to fix it using the workaround in https://github.com/stnkl/FastLEDManager/commit/54bbaf1d12c89977501e1dea95f9347ae0a954ed but it's not very clean.

prodestrian, 2022-06-26:
Thanks for posting the above workaround, I've spent close to a day trying to fix this on a project. I'm using an ESP8266 with a 60-LED Neopixel ring.

I was convinced it was a hardware issue until I installed WLED and everything worked fine (Wifi enabled, zero flickering on the LEDs). And then as soon as I went back to my FastLED project the flickering and random noise returned, despite zero changes to the hardware/pins/etc.

The above (ugly) fix seems to have completely resolved it, but I'll be following this Issue in case there's an update here so I can eventually remove the workaround.
Thanks!