FastLED/FastLED #850: FastLED.clear gets stuck when too many LEDs declared in scope.

Jclarke1029, 2019-07-28:
Hello,

I am working on a large project ~2000 LEDs, using an arduino MEGA and a mixture of WS2811 and WS2812B strips.
Up to this point I have been programming individual strip patterns, I am now trying to put all of this together into one sketch.

The issue I am having is with FastLED.clear() , if I use this to clear my LEDs at the end of a pattern then the code gets stuck somewhere, and no more LEDs light. However, If i just use  CRGB::Black on the relevant LEDS, then it is fine.

My code has this number of LEDS declared::

#define NUM_stripeO 200
#define NUM_stripeI 100
#define NUM_sting 100
#define NUM_eyebrows 30
#define NUM_eyes 16*3

// Initialize LED Arrays ----------------

CRGB stripeO[3][NUM_stripeO];
CRGB stripeI[3][NUM_stripeI];
CRGB sting[NUM_sting]; 
CRGB eyes[2][NUM_eyes];
CRGB eyebrows[2][NUM_eyebrows];

and then they are initalised as follows::

FastLED.addLeds<WS2811,2, BRG>(stripeO[1], NUM_stripeO);
  FastLED.addLeds<WS2811,3, BRG>(stripeI[1], NUM_stripeI);
  
  FastLED.addLeds<WS2811,4, BRG>(stripeO[2], NUM_stripeO);
  FastLED.addLeds<WS2811,5, BRG>(stripeI[2], NUM_stripeI);
  
  FastLED.addLeds<WS2811,6, BRG>(stripeO[3], NUM_stripeO);
  FastLED.addLeds<WS2811,7, BRG>(stripeI[4], NUM_stripeI);
  
  FastLED.addLeds<WS2811,8, BRG>(sting, NUM_sting);
  
  FastLED.addLeds<WS2812B,9, GRB>(eyes[1], NUM_eyes);
  FastLED.addLeds<WS2812B,10, GRB>(eyebrows[1], NUM_eyebrows);
  
  FastLED.addLeds<WS2812B,11, GRB>(eyes[2], NUM_eyes);
  FastLED.addLeds<WS2812B,12, GRB>(eyebrows[2], NUM_eyebrows);
.

I have created functions for each section of the show which is then called in loop. 

Below is an example of code that works. you can see at the bottom when I break the while loop, then I set all LEDs to Black manually, it is fine and cycles through like it should. 

But if I comment out that line setting to black and use FastLED.clear() instead, then it gets stuck and the LEDs cycle through once and then just stay black. 

What's even more interesting is that if i have both lines there. i.e., CRGB::Black manual and the FastLED.clear() afterwards, it still gets stuck, so it is definitely the presence of that function causing the issue.

 void BeeBase(int Speed, int cycles) {
  
  int cyclecount = 0; 
  
  while(true)
  {
  
    int dot = 0;
    

      while(true)
      {

      EVERY_N_MILLISECONDS(Dtime1*2/Speed) 
        {
           //eyebrow 

          eyebrows[1][dot] = CRGB::HotPink;
          dot=dot+1;
                  
        }
        
       FastLED.show();
       if (dot >= (NUM_eyebrows)) break;
      } 

  for (int i = 0; i <= NUM_eyebrows;i++) eyebrows[1][i] = CRGB::Black; 
  //FastLED.clear();
  cyclecount = cyclecount+1;
  if (cyclecount >= cycles) break;
  }
}

  

I think it is to do with the amount of LEDs i have declared, the FastLED.clear() is clearing everything and it is too much for it apparently. 

If i remove some of the initialisations for other strips (i.e., comment out all the FastLED.adaLEDs other than the one for this strip) and rerun then FastLED.clear() works fine., so it appears to be that it is overloaded.

Has anyone else experienced this? and does anyone know how to fix it? 
I really don't want to have to write LEDs to black manually every time.

MartyMacGyver, 2019-07-28:
Some comments:

If you edit your post and wrap your code blocks in triple backticks, it'll be more readable.

e.g.:

    ```
    code
    goes
    here
    ```

A Mega only has 8K of memory. How much free dynamic memory are you seeing free *right before* these calls? (Versus when you uploaded the sketch.)

`clear()` is [a very simple function](https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/FastLED.cpp#L108-L113), but it does end up calling `clearData()` [which iterates over the LED controllers and clears each in turn](https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/FastLED.cpp#L115-L121). This them finished with a `memset8()` to zero out the pixels in memory. (I expect this is slightly faster than zeroing memory, reading the zeroes from it, and streaming them back out.)

It's all effectively the same as setting the pixels to zero and calling `show()` yourself, but there are optimizations and to my eye the call stack will be slightly deeper if you call `clear()`.

If you're right on the edge of free memory (or already over the edge and it's just not obvious until you do this) then that would be one reason this would act up.

Try this:

Instrument your code to show free memory near the point where the problem occurs.

There is a LOT of discussion here about how to view free memory:
https://github.com/arduino/Arduino/issues/5289

The most recent comment seems to be a summary of the best ideas in the thread.

Then, reduce the number of LEDs you're allocating til the problem stops, and perhaps drop it a little more (if I'm right and this is a stack size issue, your output might be incorrect if the stack is getting corrupted, even if it's not quite enough to cause things to fully stop).

At this point (if you were overrunning the stack before) the memory readings should start making sense.

MartyMacGyver, 2019-07-28:
Actually, I'm not sure the fact that `clear()` is slightly more complex (but well-optimized) is meaningful here, except that it exercises a slightly different path in the stack. If the stack is getting corrupted all sorts of things will start going wrong, and this would just be a sign of that, rather than a cause.

When you add the code to show memory, reduce LEDs, it might actually be easier to drop the number to a very low value (like 256 or even 16) to start with, and then (assuming things work normally at that point) start increasing it and seeing how the memory usage near the call to `clear()` changes as you go.

If there's a memory leak elsewhere (another possible problem as you've only got partial code here) it should also be evident as the free memory will keep dropping as the sketch runs.

There are other complexities like memory fragmentation that might be involved, but being able to see the memory use at one (or many) points in the code will help you debug this.

Jclarke1029, 2019-07-28:
> Some comments:
> 
> If you edit your post and wrap your code blocks in triple backticks, it'll be more readable.
> 
> e.g.:
> 
> ```
> ```
> code
> goes
> here
> ```
> ```
> 
> A Mega only has 8K of memory. How much free dynamic memory are you seeing free _right before_ these calls? (Versus when you uploaded the sketch.) **EDIT: More info at the end of this post.**
> 
> `clear()` is [a very simple function](https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/FastLED.cpp#L108-L113), but it does end up calling `clearData()` [which iterates over the LED controllers and clears each in turn](https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/FastLED.cpp#L115-L121). This them finished with a `memset8()` to zero out the pixels in memory. (I expect this is slightly faster than zeroing memory, reading the zeroes from it, and streaming them back out.)
> 
> It's all effectively the same as setting the pixels to zero and calling `show()` yourself, but there are optimizations and to my eye the call stack will be slightly deeper if you call `clear()`.
> 
> If you're right on the edge of free memory (or already over the edge and it's just not obvious until you do this) then that would be one reason this would act up.
> 
> Try this:
> 
> Instrument your code to show free memory near the point where the problem occurs.
> 
> There is a LOT of discussion here about how to view free memory:
> [arduino/Arduino#5289](https://github.com/arduino/Arduino/issues/5289)
> 
> The most recent comment seems to be a summary of the best ideas in the thread.
> 
> Then, reduce the number of LEDs you're allocating til the problem stops, and perhaps drop it a little more (if I'm right and this is a stack size issue, your output might be incorrect if the stack is getting corrupted, even if it's not quite enough to cause things to fully stop).
> 
> At this point (if you were overrunning the stack before) the memory readings should start making sense.

Thanks for your reply, 

At compiling it says i am using 55% of dynamic memory, which is quite a bit of wiggle room.

just before the FastLED.clear() , the "freeMemory()" function from the post you sent ( thank you) returns 35 to the serial monitor, which i assume is a %??
 So i'm not sure that memory is the issue in that case?

MartyMacGyver, 2019-07-28:
I presume you mean [this `freeMemory()` function](https://github.com/McNeight/MemoryFree/blob/a2d60bd958f80fc3dea8479467a1309643bdbc63/MemoryFree.cpp#L39-L52)?

That's not 35 _percent_... that's 35 **bytes**!

Can you try reducing the number of pixels to see what the trend is?

Jclarke1029, 2019-07-28:
Yes, and therefore that would definitely be a problem! 
 I have now identified the issue I think!

IF you take a look at my original post above:

CRGB eyebrows[2][NUM_eyebrows];

I declare an array of 2.

FastLED.addLeds<WS2812B,9, GRB>(eyes[1], NUM_eyes);
FastLED.addLeds<WS2812B,10, GRB>(eyebrows[1], NUM_eyebrows);

FastLED.addLeds<WS2812B,11, GRB>(eyes[2], NUM_eyes);
FastLED.addLeds<WS2812B,12, GRB>(eyebrows[2], NUM_eyebrows);

but then initialize [1] and [2] instead of [0] and [1].   
Therefore i think that blocks a load of extra memory that isnt being used. 
now i've fixed my indexing to 0 ( I hate 0 indexing ) I think it's all good!!

Thank you!

closed 2019-07-28 by Jclarke1029

MartyMacGyver, 2019-07-28:
It was difficult to read... That said, with things reformatted:
```
#define NUM_stripeO 200
#define NUM_stripeI 100
#define NUM_sting 100
#define NUM_eyebrows 30
#define NUM_eyes 16*3

// Initialize LED Arrays ----------------

CRGB stripeO[3][NUM_stripeO];
CRGB stripeI[3][NUM_stripeI];
CRGB sting[NUM_sting];
CRGB eyes[2][NUM_eyes];
CRGB eyebrows[2][NUM_eyebrows];

and then they are initalised as follows::

FastLED.addLeds<WS2811,2, BRG>(stripeO[1], NUM_stripeO);
FastLED.addLeds<WS2811,3, BRG>(stripeI[1], NUM_stripeI);

FastLED.addLeds<WS2811,4, BRG>(stripeO[2], NUM_stripeO);
FastLED.addLeds<WS2811,5, BRG>(stripeI[2], NUM_stripeI);

FastLED.addLeds<WS2811,6, BRG>(stripeO[3], NUM_stripeO);
FastLED.addLeds<WS2811,7, BRG>(stripeI[4], NUM_stripeI);

FastLED.addLeds<WS2811,8, BRG>(sting, NUM_sting);

FastLED.addLeds<WS2812B,9, GRB>(eyes[1], NUM_eyes);
FastLED.addLeds<WS2812B,10, GRB>(eyebrows[1], NUM_eyebrows);

FastLED.addLeds<WS2812B,11, GRB>(eyes[2], NUM_eyes);
FastLED.addLeds<WS2812B,12, GRB>(eyebrows[2], NUM_eyebrows);
```

`stripeO` and `stripeI` are 3 elements each. Once you fix the zero indexing, there was a second error as well:

```
FastLED.addLeds<WS2811,6, BRG>(stripeO[3], NUM_stripeO);
                                   // ^^^ This is one off, as noted
FastLED.addLeds<WS2811,7, BRG>(stripeI[4], NUM_stripeI);
                                   // ^^^ This is two off!
```