FastLED/FastLED #851: #pragma message: FastLED version 3.002.010

alhaddar, 2019-07-29:
Hi
Hi
When I'm sending programs to the Arduino show me this message

In file included from C:\Users\alhaddar\Round-LED-Clock_f-home2\Round-LED-Clock_f-home2.ino:18:0:

C:\Users\alhaddar\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.010

pragma message "FastLED version 3.002.010"
                 ^
In file included from C:\Users\alhaddar\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

             from C:\Users\alhaddar\Round-LED-Clock_f-home2\Round-LED-Clock_f-home2.ino:18:
C:\Users\alhaddar\Documents\Arduino\libraries\FastLED/fastspi.h:115:23: note: #pragma message: No hardware SPI pins defined. All SPI access will default to bitbanged output

pragma message "No hardware SPI pins defined. All SPI access will default to bitbanged output"

Blink515, 2019-07-29:
Not an error. Just an informational message to let you know what version of
fastled is being used and that spi isn't enabled.

If your project is working fine you can ignore this.

alhaddar, 2019-07-29:
Thank you for your response
Unfortunately my project did not work for this reason Send this problem until I get a solution
Regards

henrygab, 2019-07-31:
@alhaddar
If you are uncomfortable with this message appearing, you can silence it yourself:
https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/FastLED.h#L12-L14

Therefore, as an example, you could make the following one-line change to silence it:

```c
#define FASTLED_INTERNAL // add this before including FastLED.h
#include <FastLED.h> // this should already exist in your project
```

alhaddar, 2019-08-01:
Thank you for your help and appreciate this help

Regards

henrygab, 2019-08-02:
@alhaddar Glad to be able to help.  If this solves your problem, please click the "close" button on this thread, so it doesn't show up as an active issue?  Thanks!

closed 2019-08-12 by focalintent

egnor, 2021-07-30:
I do understand your intent, however, literally nobody else in the known universe uses `#pragma message` to report version (or at all, really) in widely used code, because people don't like junk in successful compiler output. If every library I used did this, my compiler output would be a mountain of garbage; FastLED is a very nice library but it's not _that_ special compared to the others.

Please consider _at least_ adding an `#if` to suppress this message? Currently, `#define FASTLED_INTERNAL` will suppress it, and an increasing number of people are defining that, but surely that seems like a poor habit given the name? Something like `#define FASTLED_USE_PRAGMA 0` would be clearer.