FastLED/FastLED #852: Fastled 3.2.6-dma fork by EmteZogaf conflicts with Esp8266 v2.5.2

tihoangyeudau, 2019-07-29:
[Fastled 3.2.6-dma fork by EmteZogaf](https://github.com/EmteZogaf/FastLED/tree/dma) work well with esp8266 v2.5.0, but with v2.5.1 and v2.5.2, there are conflicts:

`C:\Users\tuant\Documents\Arduino\libraries\FastLED-3.2.10/platforms/esp/8266/clockless_esp8266_dma.h:39:124: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);

In file included from C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/esp8266_peri.h:25:0,

                 from C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:38,

                 from sketch\esp8266-fastled-webserver-2.ino.cpp:1:

C:\Users\tuant\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/esp8266_undocumented.h:9:13:  error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here
 extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);
`
can anyone help me to fix it? i have to use it for RGBW

tihoangyeudau, 2019-07-30:
i fixed it by myself!

closed 2019-07-30 by tihoangyeudau

CWempe, 2019-11-30:
Had the same issue.

Solved this by removing the line in this library, because it exists in the later esp library.

https://github.com/CWempe/FastLED/commit/19cb098acd97bcdd72306ce43af93742b974fb99