FastLED/FastLED #853: random8(int lim) will truncate the answer if r*lim is > 255

ssilverman, 2019-07-29:
The `r*lim` multiplication stays in type `uint8_t` and will truncate if the result would have been > 255.

https://github.com/FastLED/FastLED/blob/3190e9673f94f3bb0d654fbd67d4a298d5f44896/lib8tion/random8.h#L40

kriegsman, 2019-07-30:
Are you seeing this happen in practice, and if so, which microcontroller and compiler version are you using?  Do you have specific values that are triggering a wrong result?

I'm asking for the details because I believe that type of the result of an otherwise unqualified multiplication operation ('*') is 'int' or 'unsigned int'.  That means that on standard AVR/Arduino, we'd get int16_t or uint16_t for the result type.

I haven't seen the truncation error that you describe, but I'm also willing to believe that some combination of chip and compiler version might be producing code that truncates the result.  So, let me know where you're seeing the problem in practice and/or paste a short test case in here, and I'll investigate and fix it if there's a problem on one particular platform.  

I'll give it a few days for you to provide the details of how to reproduce the bug, but if you're not seeing any problem in practice, I'll close the ticket after a little while.

kriegsman, 2019-07-30:
At the assembly level, on AVR, the 8bit * 8bit multiply is one instruction that yields a 16-bit result.  The 8-bit shift-right in random8 makes the compiler select the high eight bits of the sixteen bit result, and return them.  Here's a link to what the source compiles down into on the 'compiler explorer' web site: https://godbolt.org/z/6t7TE4

And again, this was compiled for AVR/ATmega/Ardunino.  If you're seeing a truncation problem on a a non-ATmega platform, or with a different compiler, please let me know.  We rely on the 8-bit * 8-bit = 16-bit all over the place in FastLED and if you've found some place that it's not working, I'd love to understand and fix it.

ssilverman, 2019-07-31:
Well, this is embarrassing. I often don't sound so definite in my issue titles, but this time, I thought I was sure. After reading the C++ spec _again_ (https://en.cppreference.com/w/cpp/language/operator_arithmetic, https://en.cppreference.com/w/cpp/language/implicit_conversion#Integral_promotion), I've decided that I really _don't_ yet understand how multiplicative overflow works.

Closing with head down...

closed 2019-07-31 by ssilverman

kriegsman, 2019-07-31:
No worries, and thanks for bothering to raise the issue when you thought there might be one!

As a thank you gift, here's a link to a blog post that I made when I discovered some int-type-width surprises on AVR/Arduino: https://blog.kriegsman.org/2013/12/01/optimizing-10-lines-of-arduino-code-youre-wrong-so-am-i/

ssilverman, 2019-07-31:
Am I correct about the operands automatically being converted to `int`, per that sentence I mention above (“In particular, arithmetic operators do not accept types smaller than int as arguments.”) from that second link?