FastLED/FastLED #856: nRF52 - First call to showPixels() with neopixel results in wrong color

henrygab, 2019-08-03:
Reported via #840 

Repro is trivial:
```c++
#define FASTLED_INTERNAL
#include <FastLED.h>
#define NUM_LEDS 48
CRGB leds[NUM_LEDS];
// LED_RED      for built-in red LED   == 3
// LED_BLUE     for built-in blue LED  == 4
// PIN_NEOPIXEL for built-in NEOPIXEL  == 8
void setup() {
  delay(2000);
  pinMode(PIN_NEOPIXEL, OUTPUT);
  pinMode(LED_RED,      OUTPUT);
  pinMode(LED_BLUE,     OUTPUT);
  FastLED.addLeds<NEOPIXEL, PIN_NEOPIXEL>(leds, NUM_LEDS);
  FastLED.setBrightness(10);
  fill_solid(leds, NUM_LEDS, CRGB::Blue);
  FastLED.show();
}

void loop() {
  static int i = 0;
  for (int j = 0; j < 0x200; j++) {
    fill_solid(leds, NUM_LEDS, CRGB::Blue);
    FastLED.show();
    fill_solid(leds, NUM_LEDS, CRGB::Red);
    FastLED.show();
  }
  i++;
  digitalWrite(LED_RED, (i % 2) ? HIGH : LOW); // blink LED
}
```

henrygab, 2019-08-10:
@masonevans -- Would be willing to validate a potential fix for #856 (first LED wrong color first time only)?  See
https://github.com/henrygab/FastLED/commit/ded19a59cca36a05dcf93cd2095ce4a6f68e15e2.

The potential fix is to add the following four lines (two of which are comments) to file [platforms/arm/nrf52/clockless_arm_nrf52.h](https://github.com/henrygab/FastLED/blob/ded19a59cca36a05dcf93cd2095ce4a6f68e15e2/platforms/arm/nrf52/clockless_arm_nrf52.h#L159-L162) at function init():
```C++
    // to avoid pin initialization from causing first LED to have invalid color,
    // call mWait.mark() to ensure data latches before color data gets sent.
    startPwmPlayback_InitializePinState();
    mWait.mark();
```
While I couldn't get my scope connected to pin 8 to prove the cause,
it appears to prevent any the first LED color issue, without affecting
later calls to `showPixels()`.

Can you please let me know, whether you don't test it, or the results you get when testing it?

Thanks!

henrygab, 2019-08-10:
OK, I confirmed this is caused by a lack of setting the pin to the proper state at startup.

You can fix this in your sketch by ensuring (for non-inverted chipsets) that you set the pin  
```c++
  digitalWrite(PIN_NEOPIXEL, LOW);
  delay(100);
```

I will also push the preventative measure above.

masonevans, 2019-08-12:
@henrygab thanks for the update on this one, I am out of town until Friday but I will check on this fix when I get back home

PR #860 (2019-08-13): NRF52 -- Prevent pin initialization affecting clockless LED data
Avoid first-time pin initialization from affecting showLeds() for clockless chipsets.

Fixes #856.

If a sketch does not initialize pins to low state (or high when _FLIP=1), 
the first call to showLeds() may intialize pin state, which clockless LEDs
may treat as part of a data stream.

Avoid this by also initializing the pin state in Init(), and
call mWait.mark() to ensure LEDs will not consider this part of the
LED data.
commit 44325974 (2019-08-13): Fixes #856. (#860)

Initialization of pins may cause change in pin output.
Fix may be to initialize th epin state in Init(), and
call mWait.mark() to prevent showPixels() from sending
actual LED data to the strand until any potential
prior data had time to latch.
closed 2019-08-13 by focalintent