FastLED/FastLED #858: ATMega4809 support?

Mnjongen, 2019-08-06:
Is the ATMega4809 (Arduino nano every) going to be supported? I noticed that the Arduino nano IOT was already added but not the nano every. 

I tried adding it myself but I couldn't understand the fastpin_avr.h file.

jvcop, 2019-08-07:
Support for the Arduino Nano Every would be wonderful. I'm willing to contribute, but I'm new to these kinds of things.

So far, I tried reusing the pin definitions of the ATmega328P (Nano v3) as I think the Nano Every can be used as a drop-in replacement. I might be wrong, though. In any case, I ran into a number of follow-up problems that I didn't have time to investigate yet.

Mnjongen, 2019-08-07:
I have been trying some things (example: making it look like it's a nano) but still couldn't get it to work. I'll just have to wait for someone with a solution.

focalintent, 2019-08-12:
I have a nano every on order and once it gets here I can start looking at getting it supported - I don’t think it is just going to be a new set of pin definitions.

focalintent, 2019-08-15:
The ATMega4809 is different enough from the rest of the AVR based boards that FastLED supports that it will effectively be a full port (ARCH_ARDUINO_AVRMEGA vs. ARCH_ARDUINO_AVR).  The pin work is done on a branch, but it can't make use of the existing avr asm code for things like the WS2812 output and such, there are enough differences already.  I'm going to be re-writing the asm code for both avr and m0 already for the RGBW support - I'll roll in updating the asm for the ATmega4809 into that process, but probably not before then.

jvcop, 2019-08-15:
@focalintent Thanks for the update!

Speaking of WS2812, would the WS2801 be supported as well?

focalintent, 2019-08-19:
Yes, although not with hardware SPI - you can check out the branch here  https://github.com/FastLED/FastLED/tree/avrmega -- but I won't merge it back into master until I have a solution for WS281x style chips working.

hiduino, 2019-08-29:
Using the flat register names you could use the existing fastpin_avr.h file for the port and pin definitions.  I've use this to get the ATtiny series 0/1 and ATmega series 0 to work with DotStar LEDs.

`#if !defined(__AVR_ATtinyxy7__) && !defined(__AVR_ATtinyxy6__) && !defined(__AVR_ATtinyxy4__) && !defined(__AVR_ATtinyxy2__) && !defined(__AVR_ATmega4809__)

#define _FL_IO(L,C) _RD8(DDR ## L); _RD8(PORT ## L); _RD8(PIN ## L); _FL_DEFINE_PORT3(L, C, _R(PORT ## L));
#define _FL_DEFPIN(_PIN, BIT, L) template<> class FastPin<_PIN> : public _AVRPIN<_PIN, 1<<BIT, _R(PORT ## L), _R(DDR ## L), _R(PIN ## L)> {};

#else
// Register name equivalent (using flat names) DDRA as PORTA_DIR,  PORTA as PORTA_OUT,  PINA as PORTA_IN
#define _FL_IO(L,C) _RD8(PORT ## L ## _DIR); _RD8(PORT ## L ## _OUT); _RD8(PORT ## L ## _IN); _FL_DEFINE_PORT3(L, C, _R(PORT ## L ## _OUT));
#define _FL_DEFPIN(_PIN, BIT, L) template<> class FastPin<_PIN> : public _AVRPIN<_PIN, 1<<BIT, _R(PORT ## L ## _OUT), _R(PORT ## L ## _DIR), _R(PORT ## L ## _IN)> {};

#endif
`

Rydogdav, 2019-11-02:
Any further progress on this? I'd really like to be able to use my WS2812Bs. Thanks!

graysonmendenhall, 2020-02-06:
Has there been any more progress on this issue or should I just revert to using a regular Nano?

kriegsman, 2020-02-06:
Not supported yet but we’re starting to collect relevant info over on https://github.com/FastLED/FastLED/issues/716
Would love to have help if you have some expertise with the ATmega4809 pin/port mappings, too

ngyl88, 2024-09-18:
@Mnjongen, @kriegsman or other repo maintainers:
I suggest we closed this issue as https://github.com/FastLED/FastLED/issues/716 covers both Arduino Uno Wifi Rev. 2, Arduino Nano Every. 

I believe this issue is mostly around Arduino Nano Every, which is the subset of ATmega changes over the recent years.
- https://github.com/FastLED/FastLED/pulls?q=is%3Apr+Nano+Every

We should be able to group and track similar and more recent issue with
- https://github.com/FastLED/FastLED/issues?q=atmega4809
- https://github.com/FastLED/FastLED/issues?q=%22arduino+nano+every

Mnjongen, 2024-09-18:
I will close the issue as suggested. It seems some development has been done on the every, and newer issues are probably more relevant and detailed.

closed 2024-09-18 by Mnjongen