FastLED/FastLED #861: Power management seriously broken with OctoWS2811 controller

iceboundflame, 2019-08-11:
Hi, FastLED is severely underestimating the power usage when using the OctoWS2811 controller, potentially damaging projects.

The OctoWS2811 controller is initialized by providing the length of one channel only, and it assumes you have 8x channels with that many pixels. This length of the single channel is stored as the controller's `size()`.

The power computation used for `calculate_max_brightness_for_power_mW` only checks the first `size()` pixels in the LED buffer -- thus, it doesn't correctly compute the power usage.

For example: When the first `size()` pixels are black, then it thinks the power demanded is simply the 'off' current of those first pixels. The rest of the pixels could be on full white.

```
CRGB leds[90*8];  // leds is 720 long
void setup() {
  FastLED.addLeds<OCTOWS2811>(leds, 90);  // controller only has "size" of 90
}
```

kriegsman, 2019-08-11:
Thanks for the report -- much appreciated, and oops, that's a real bug.

I assume you already have the simple work-around for your own project: reduce the 'max power' by a factor of 8 (or however many strips you actually have).

I'll take a look and see if there's a quick way to address this... but I see some potential challenge if the _actual_ number of LEDs _is never provided to FastLED_.  I'll take a look, and thanks again.

focalintent, 2019-08-11:
It turns out I had already thought about this with the block controllers - just missed putting it into the OctoWS2811 controller - but if you look at https://github.com/FastLED/FastLED/blob/master/platforms/arm/k20/clockless_block_arm_k20.h#L31 - from the non parallel  controller there's this function:

```
virtual int size() { return CLEDController::size() * LANES; }
```

adding this to the OctoWS2811 controller should be enough to ensure that the right size is returned and used by the power function for navigating/adding up the led data.

focalintent, 2019-08-11:
Checked in a fix for this - it'll be in  the next spun up  release (or you can just grab master@HEAD from GitHub  right now)

iceboundflame, 2019-08-11:
Thank you both!

@kriegsman Unfortunately dividing by 8 doesn't quite work because power consumption is only tallied for the first 1/8 of the pixels, which may be set to a different color than the rest of the array. Thus it could be significantly more or less than 8x off.

@focalintent Awesome, thanks for the quick fix -- I'll give it a try.

iceboundflame, 2019-08-12:
@focalintent Confirmed the fix worked, thanks!

closed 2019-08-12 by iceboundflame

focalintent, 2019-08-12:
Great!  I'm working on finishing up support for the Teensy 4.0 and once that's done I'll spin the release.