FastLED/FastLED #865: still doesn't compile with Atmega32

Master811129, 2019-08-14:
hi there.
after downloading the new version of the FastLED library (3.3) still i can't compile with atmega32.
the developer said in this topic #598: this will add in FastLED 3.2 but after releasing FastLED 3.3 it doesn't support atmega32 MCU.
thanks

focalintent, 2019-08-14:
What board uses the atmega32 and what are you selecting in the Arduino environment to build for it?  It is likely that I still need pin definitions for it - as I don’t have any atmega32 based boards to test myself.

focalintent, 2019-08-14:
Ah - I see what happened - I asked you questions for clarification - you never answered them but someone else did and I went through and put in what was needed based on those answers.  So - I need you to give me the answers to my questions if you want this actually fixed.

Master811129, 2019-08-14:
> What board uses the atmega32 and what are you selecting in the Arduino environment to build for it? It is likely that I still need pin definitions for it - as I don’t have any atmega32 based boards to test myself.

the board is this:
https://github.com/MCUdude/MightyCore
this is an unofficial arduino core for supporting more MCUs like the powerful atmega32.(this is cheaper than the atmega328 an my country so this is why i gonna use this microcontroller.)
i'm using arduino 1.8.9

here is the errors
`In file included from C:\Users\Master\Documents\Arduino\libraries\FastLED\examples\Blink\Blink.ino:1:0:

C:\Users\Master\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.000

 #    pragma message "FastLED version 3.003.000"

                     ^

In file included from C:\Users\Master\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Master\Documents\Arduino\libraries\FastLED\examples\Blink\Blink.ino:1:

C:\Users\Master\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\Master\Documents\Arduino\libraries\FastLED/FastLED.h:48:0,

                 from C:\Users\Master\Documents\Arduino\libraries\FastLED\examples\Blink\Blink.ino:1:

C:\Users\Master\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<3u>':

C:\Users\Master\Documents\Arduino\libraries\FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<3u, 4, 10, 6, (EOrder)66u, 0, false, 10>'

C:\Users\Master\Documents\Arduino\libraries\FastLED/chipsets.h:478:7:   required from 'class WS2812Controller800Khz<3u, (EOrder)66u>'

C:\Users\Master\Documents\Arduino\libraries\FastLED/FastLED.h:92:34:   required from 'class NEOPIXEL<3u>'

C:\Users\Master\Documents\Arduino\libraries\FastLED/FastLED.h:314:28:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 3u]'

C:\Users\Master\Documents\Arduino\libraries\FastLED\examples\Blink\Blink.ino:23:56:   required from here

C:\Users\Master\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

Using library FastLED at version 3.3.0 in folder: C:\Users\Master\Documents\Arduino\libraries\FastLED 
exit status 1
Error compiling for board ATmega32.`

i also compile with the Adafruit_NeoPixel library and it compiles and works like a charm. but this library cannot compile with this MCU.

focalintent, 2019-08-14:
This is because of the way FastLED does low level pin access - but it means that definitions for the pins need to be explicitly defined - you can see example definitions in here - https://github.com/FastLED/FastLED/blob/master/platforms/avr/fastpin_avr.h -- without those definitions, then there's no mapping from the pins to the GPIO port registers, which is needed by FastLED at compile time.    I'll take a look at that repo and see if I can pull out the pin/port numbers that will be needed for this board to work.

Master811129, 2019-08-16:
> This is because of the way FastLED does low level pin access - but it means that definitions for the pins need to be explicitly defined - you can see example definitions in here - [...]

bro I'm not a professional C++ developer like you but i hope this will help you:
http://uupload.ir/files/lf70_68747470733a2f2f692e696d6775722e636f6d2f7a424e506335622e706e67.png

it seems the atmega1284 and Atmega32A has same pin definition. the Atmega1284 compiles with FastLED library.
maybe we can add this MCU in this line of fastpin_avr.h
`#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)`

Master811129, 2019-08-17:
> This is because of the way FastLED does low level pin access - but it means that definitions for the pins need to be explicitly defined - you can see example definitions in here - [...]

Do you have any idea.👆🏼👆🏼👆🏼

focalintent, 2019-08-17:
Try it out!   I won’t be able to spend time with this side of things until tomorrow or Monday.

Master811129, 2019-08-20:
> Try it out! I won’t be able to spend time with this side of things until tomorrow or Monday.
you updated FastLED library and still....Ah....  Is there any chance FastLED will support this MCU?

i can't find fastpin_avr.h.

focalintent, 2019-08-20:
It’s in the folder FastLED/platforms/avr

And yes, I updated with an emergency fix because an already supported platform was failing to build for people, and I was able to do most of what I needed for that from my phone/iPad.  For this one I need some time with my laptop - I have a full time job that is not FastLED and sometimes I don’t have a lot of time to put on this.

focalintent, 2019-08-26:
It should now with 3.3.2

closed 2019-08-26 by focalintent

Master811129, 2019-08-26:
> It should now with 3.3.2

yes finally
thank you sir.
let's test it out

commit 6110d35f (2019-08-26): Fix #865 - use the atmega1284/644 definitions for atmega32 as well