FastLED/FastLED #868: compile errors for Teensy LC

manitou48, 2019-08-18:
compile error for blink example on Teensy LC using FastLED version 3.003.000, see
https://forum.pjrc.com/threads/57153-FastLED-not-compiling-on-Teensy-LC-SysTick-not-defined

marmilicious, 2019-08-18:
I noticed the compile info says:
 Multiple libraries were found for "FastLED.h"
 Used: /home/henryt/Arduino/libraries/FastLED
 Not used: /home/henryt/arduino-1.8.9/hardware/teensy/avr/libraries/FastLED

You should (re)move one of those so the compiler only finds one version of the library.  You don't want duplicate libraries.  Try removing the one under teensy/avr/libraries.

manitou48, 2019-08-18:
I did the test myself just now. removed all old FastLED libs, and cloned your latest from github.  Compile of blink example still fails for Teensy LC:

 #    pragma message "FastLED version 3.003.000"
                     ^
In file included from /home/dunigan/sketchbook/libraries/FastLED/platforms/arm/kl26/clockless_arm_kl26.h:4:0,
                 from /home/dunigan/sketchbook/libraries/FastLED/platforms/arm/kl26/fastled_arm_kl26.h:7,
                 from /home/dunigan/sketchbook/libraries/FastLED/platforms.h:20,
                 from /home/dunigan/sketchbook/libraries/FastLED/FastLED.h:52,
                 from /home/dunigan/sketchbook/libraries/FastLED/examples/Blink/Blink.ino:1:
/home/dunigan/sketchbook/libraries/FastLED/platforms/arm/kl26/../common/m0clockless.h: In function 'int showLedData(volatile uint32_t*, uint32_t, const uint8_t*, uint32_t, M0ClocklessData*)':
/home/dunigan/sketchbook/libraries/FastLED/platforms/arm/kl26/../common/m0clockless.h:316:40: error: 'SysTick' was not declared in this scope
       uint32_t ticksBeforeInterrupts = SysTick->VAL;
                                        ^
/home/dunigan/sketchbook/libraries/FastLED/platforms/arm/kl26/../common/m0clockless.h:326:36: error: 'VARIANT_MCK' was not declared in this scope
       const uint32_t kTicksPerMs = VARIANT_MCK / 1000;
                                    ^
Error compiling for board Teensy LC.

focalintent, 2019-08-18:
Yeah - I discovered this the other night - someone made changes to the arm m0 clockless cose that’s breaking this - until I get a fix checked in (there’s one on a branch for other work, need to merge it over) put:

#define FASTLED_ALLOW_INTERRUPTS 0 

Before the include FastLED.h (their change was a hack to allow it on more platforms - but the teensy LC doesn’t work with it - but it defines allow interrupts as 1 even though it never used it before)

focalintent, 2019-08-18:
Spun up FastLED 3.3.1 w/the fix for this in place.

commit 3e962863 (2019-08-18): Fix #868 for Teensy LC
closed 2019-08-18 by focalintent