FastLED/FastLED #870: ESP8266 and APA102 compile error

gunnwo, 2019-08-21:
I used the DemoReel100 example and ran into a problem with compilation. Code adaptions done so far:

#define FASTLED_ESP8266_RAW_PIN_ORDER_
#include <FastLED.h>
...
#define DATA_PIN    3
#define CLK_PIN   4
#define LED_TYPE    APA102
#define COLOR_ORDER BGR
#define NUM_LEDS    64
...
FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
...

Compiling with Arduino Studio 1.8.9, ESPboard NodeMCU 0.9 with boad info esp8266 v2.5.2, FastLED 3.3.1 fails with an error in FastLED. 

```
In file included from C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/FastLED.h:44:0,

                 from D:\temp\DemoReel100\DemoReel100.ino:3:

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: error: template instantiation depth exceeds maximum of 900 (use -ftemplate-depth= to increase the maximum) substituting 'template<int CYCLES> void delaycycles() [with int CYCLES = 1073740926]'

  FL_NOP; delaycycles<CYCLES-1>();
                                ^
C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32:   recursively required from 'void delaycycles() [with int CYCLES = 1073741822]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32:   required from 'void delaycycles() [with int CYCLES = 1073741823]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastspi_bitbang.h:177:52:   required from 'static void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeBit(uint8_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t) [with unsigned char BIT = 0u; unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; unsigned int SPI_SPEED = 5u; uint8_t = unsigned char; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t = unsigned int; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t = unsigned int]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastspi_bitbang.h:345:14:   required from 'void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writePixels(PixelController<RGB_ORDER>) [with unsigned char FLAGS = 128u; D = DATA_NOP; EOrder RGB_ORDER = (EOrder)136u; unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; unsigned int SPI_SPEED = 5u]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/chipsets.h:426:3:   required from 'void SM16716Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; EOrder RGB_ORDER = (EOrder)136u; unsigned int SPI_SPEED = 5u]'

D:\temp\DemoReel100\DemoReel100.ino:127:1:   required from here

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: error: no matching function for call to 'delaycycles()'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: note: candidate is:

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:99:65: note: template<int CYCLES> void delaycycles()

 template<int CYCLES> __attribute__((always_inline)) inline void delaycycles() {
                                                                 ^
C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:99:65: note:   substitution of deduced template arguments resulted in errors seen above

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: error: template instantiation depth exceeds maximum of 900 (use -ftemplate-depth= to increase the maximum) substituting 'template<int CYCLES> void delaycycles() [with int CYCLES = 2147482749]'

  FL_NOP; delaycycles<CYCLES-1>();
                                ^
C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32:   recursively required from 'void delaycycles() [with int CYCLES = 2147483645]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32:   required from 'void delaycycles() [with int CYCLES = 2147483646]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastspi_bitbang.h:176:52:   required from 'static void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeBit(uint8_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t) [with unsigned char BIT = 0u; unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; unsigned int SPI_SPEED = 3u; uint8_t = unsigned char; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t = unsigned int; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t = unsigned int]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastspi_bitbang.h:345:14:   required from 'void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writePixels(PixelController<RGB_ORDER>) [with unsigned char FLAGS = 0u; D = DATA_NOP; EOrder RGB_ORDER = (EOrder)136u; unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; unsigned int SPI_SPEED = 3u]'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/chipsets.h:138:3:   required from 'void WS2801Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 3u; unsigned char CLOCK_PIN = 4u; EOrder RGB_ORDER = (EOrder)136u; unsigned int SPI_SPEED = 3u]'

D:\temp\DemoReel100\DemoReel100.ino:127:1:   required from here

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: error: no matching function for call to 'delaycycles()'

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:101:32: note: candidate is:

C:\UserData\XXXXXX\Documents\Arduino\libraries\FastLED/fastled_delay.h:99:65: note: template<int CYCLES> void delaycycles()

 template<int CYCLES> __attribute__((always_inline)) inline void delaycycles() {
```

Note: compiling the example as it is works fine. There must be a problem with APA102 I assume.
Any input welcome.

Thanks

focalintent, 2019-08-21:
Huh - there might be a bug with the bit banging SPI code - I’ll have to take a look at it when I get home tonight

electronix, 2019-08-21:
Same problem when using the blink example and WS2801 chips (`FastLED.addLeds<WS2801, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);`)

gunnwo, 2019-08-22:
I switched to older versions of FastLED. V3.2.10 is compiling again whereas all 2 newer versions 3.3.1 and 3.3.0 are issuing those compilation error.

BigPoit, 2019-08-22:
Same problem here with SK9822
Arduino 1.8.9
Fastled 3.3.1
NodeMCU 1.0

focalintent, 2019-08-23:
Fix checked in on master - which you can grab from GitHub, I'll spin a release later this weekend after I get a few other fixes in.

closed 2019-08-23 by focalintent

electronix, 2019-08-28:
Thanks for fixing. Works for me 👍

commit 3c5484c3 (2019-08-23): FIx #870 -- make sure that the spi clock times, now that the template parameter is a 32-bit integer, do not wrap around to super huge numbers