FastLED/FastLED #873: compilation error using latest version

GianniDPC, 2019-08-26:
`fastpin_arm_stm32.h:61:59: error: expected constructor, destructor, or type conversion before '(' token
   #define _FL_IO(L,C) _RD32(GPIO ## L);  __FL_DEFINE_PORT3(L, C, _R(GPIO ## L));`

GianniDPC, 2019-08-26:
There is another I believe in fastpin.h number 259 and 264.

shouldn't `static const void *portAddr() { return (void*)&__t_baseType::r(); } };` be `static const void *portAddr() { return (void*)__t_baseType::r(); } };` ?

Otherwise it doesn't compile and throws following error:

```
fastpin.h:264:74: error: lvalue required as unary '&' operand
           static const void *portAddr() { return (void*)&__t_baseType::r(); } };
```

GianniDPC, 2019-08-26:
Ok, take your time.

GianniDPC, 2019-08-27:
> Check out this branch -- https://github.com/FastLED/FastLED/tree/bug873 -- the changes are a bit more extensive, so I didn't want to put them onto master until I had a better idea as to whether or not they worked out

I think
 ```
typedef volatile uint32_t * port_ptr_t;
typedef uint32_t port_t;

should be outside the template otherwise it isn't accessible outside (starting line 50).

Then the compilation works untill I get the same error message as before:
```
C:\Users\Gianni\Documents\Arduino\libraries\FastLED/platforms/arm/stm32/fastpin_arm_stm32.h: In static member function 'static const void* __FL_PORT_INFO<0>::portAddr()':
C:\Users\Gianni\Documents\Arduino\libraries\FastLED/fastpin.h:264:74: error: lvalue required as unary '&' operand
           static const void *portAddr() { return (void*)&__t_baseType::r(); } };
```

sallevan, 2019-12-16:
> Check out this branch -- https://github.com/FastLED/FastLED/tree/bug873 -- the changes are a bit more extensive, so I didn't want to put them onto master until I had a better idea as to whether or not they worked out

Unfortunately it does not work either -- but o.k, it does not work some other way. :)

`Documents\Arduino\libraries\FastLED/platforms/arm/stm32/fastpin_arm_stm32.h:44:85: error: incomplete type '__gen_struct_GPIOA_BASE' used in nested name specifier
   inline static port_ptr_t port() __attribute__ ((always_inline)) { return &_GPIO::x()->ODR; }`

BTW, in the following code:

`#ifdef GPIOA
_FL_IO(A,0);
#endif
#ifdef GPIOB
_FL_IO(B,1);
#endif
#ifdef GPIOC
_FL_IO(C,2);
#endif
#ifdef GPIOD
_FL_IO(D,3);
#endif
#ifdef GPIOE
_FL_IO(E,4);
#endif
#ifdef GPIOF
_FL_IO(F,5);
#endif
#ifdef GPIOG
_FL_IO(G,6);
#endif`
nothing will be defined. I have STM32duino.com core (old Maple port, not mainstream), and there are no traces of GPIOA..GPIOG defined anywhere.