FastLED/FastLED #882: Arduino Mkr wifi 1010

julesmoretti, 2019-09-03:
Hello there:

Is there a way to make it work with an Arduino Mkr Wifi 1010?

Right now I get:
```
In file included from /Users/jules/sketchbook/FastLED-simple/FastLED-simple.ino:1:0:
/Users/jules/sketchbook/libraries/FastLED/FastLED.h:585:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
 #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
  ^~~~~~~
In file included from /Users/jules/sketchbook/FastLED-simple/FastLED-simple.ino:1:0:
/Users/jules/sketchbook/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jules/sketchbook/libraries/FastLED/FastLED.h:65:0,
                 from /Users/jules/sketchbook/FastLED-simple/FastLED-simple.ino:1:
/Users/jules/sketchbook/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jules/sketchbook/libraries/FastLED/FastLED.h:48:0,
                 from /Users/jules/sketchbook/FastLED-simple/FastLED-simple.ino:1:
/Users/jules/sketchbook/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<6>':
/Users/jules/sketchbook/libraries/FastLED/fastpin.h:239:29:   required from 'class FastPinBB<6>'
/Users/jules/sketchbook/libraries/FastLED/platforms/arm/d21/clockless_arm_d21.h:10:52:   required from 'class ClocklessController<6, 12, 30, 18, (EOrder)66, 0, false, 50>'
/Users/jules/sketchbook/libraries/FastLED/chipsets.h:582:7:   required from 'class WS2812Controller800Khz<6, (EOrder)66>'
/Users/jules/sketchbook/libraries/FastLED/FastLED.h:92:34:   required from 'class NEOPIXEL<6>'
/Users/jules/sketchbook/libraries/FastLED/FastLED.h:314:28:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 6]'
/Users/jules/sketchbook/FastLED-simple/FastLED-simple.ino:6:46:   required from here
/Users/jules/sketchbook/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino MKR WiFi 1010.
```

Thank you.

anoadragon453, 2020-01-19:
It can work! In `/path/to/Arduino/libraries/FastLED-master/platforms/arm/d21/fastpin_arm_d21.h` you just need to change the line:

```
#elif defined(ARDUINO_SAMD_MKR1000)
```

to

```
#elif defined(ARDUINO_SAMD_MKR1000) || defined(ARDUINO_SAMD_MKRWIFI1010)
```

And then it should Verify/Upload fine :)

Source: https://github.com/FastLED/FastLED/issues/414#issuecomment-444734818

p.s. And for those trying to figure out what value to put for `LED_PIN`, the DigitalPins here are what you want: 

[image]

Source: https://store.arduino.cc/arduino-mkr-wifi-1010

julesmoretti, 2020-01-21:
Now that’s a pretty amazing response! Thank you!!!!

closed 2020-01-21 by julesmoretti

anoadragon453, 2020-01-22:
And as of the next release patching should no longer be necessary: https://github.com/FastLED/FastLED/pull/956