FastLED/FastLED #883: Issue with Fastled on STM32 (Bluepill, Roger Clark build STM Implementarion)

ahuxtable, 2019-09-05:
I am trying to get this working for a few days now.  I could not get it to work at all with the official STM release (does not recognize the STM environment, so tries to use avr/io.h and fails despite #define __STM31F1__). 

Moved to Roger Clark version which allows me to get as far as the pin definition but it still fails:

FastLED/platforms/arm/stm32/fastpin_arm_stm32.h:52:86: error: incomplete type '__gen_struct_GPIOC' used in nested name specifier

I have looked over the code for the library but it's a bit much for me to understand why it might be failing.

I have specified the Pin number (3 in this case), or 'PC14', tried different pins on different ports etc but get the same error

Roger Clark STM32 Implementation on a STM32F103CT (BluePill), Arduino IDE 1.8.9, FastLed 3.3.2.

Abhyudaya98, 2019-09-06:
I am having a simillar problem - getting WS2812B working with Blue Pill (stm32f103) with arduino IDE. I got them to work with my Arduino Mega flawlessly. I get a bunch of red errors in the IDE when i run the same code with the appropriate pin for the blue pill. There are many posts regarding this. I have tried a couple solutions. None have worked so far. If there is a known fix - please let me know. 
[image]

Abhyudaya98, 2019-09-06:
Ok ... I found a quick fix for getting the blue pill to work with the fastLED library. I set the variant to be STM32F103CB (20k RAM, 128k flash) instead of (20k RAM, 64k flash) in the tools menu of the Arduino IDE.

avinashsaraf, 2019-10-30:
> Ok ... I found a quick fix for getting the blue pill to work with the fastLED library. I set the variant to be STM32F103CB (20k RAM, 128k flash) instead of (20k RAM, 64k flash) in the tools menu of the Arduino IDE.

This solution did not work for me. I am getting the error same as mentioned by @ahuxtable : incomplete type '__gen_struct_GPIOB' used in nested name specifier

metalstarlight, 2019-11-04:
> FastLED/platforms/arm/stm32/fastpin_arm_stm32.h:52:86: error: incomplete type '__gen_struct_GPIOC' used in nested name specifier

Copy files from https://github.com/metalstarlight/stm32/tree/master/FastLed-stm32duino into "\Arduino\libraries\FastLED\platforms\arm\stm32".

sallevan, 2019-12-18:
> Copy files from https://github.com/metalstarlight/stm32/tree/master/FastLed-stm32duino into "\Arduino\libraries\FastLED\platforms\arm\stm32".
And it works! Practically. :)

But dithering doesn't work for some unknown reason, i.e. using brightness different from 255 produces incorrect colors.

zackees, 2023-11-04:
Any way you could open a pull request with these fixes?

AussieMakerGeek, 2023-11-08:
> Any way you could open a pull request with these fixes?

I'm not sure this is worthy of it just yet - If there is still functionality issues, we should not be doing a pull request for it.

Can anyone confirm that this is still even an issue?  I raised this 4 years ago now and even the above referenced issue is almost a year old.

sallevan, 2023-11-08:
> Can anyone confirm that this is still even an issue?

It depends.

If you raise a question as "could the code be compiled on STM32", this is not an issue, "official" STM32 Core from ST Microelectronics has correct definitions for types (while having incorrect definition for F_CPU from the point of view of FastLED).
But when it comes to compile the code precisely with "original" Roger Clark core, it won't do.

AussieMakerGeek, 2023-11-10:
Well I would argue that when I raised the original issue, that was with the Official STM32 Core for Arduino which has dome a long way since then, and therefore this issue should be closed.  Although it's still under development, I would say the Roger Clark version of STM32 for Arduino is alternative to the official ST one. If someone has an issue with the RC version, they should raise it.

closed 2023-11-10 by AussieMakerGeek