FastLED/FastLED #885: Issue with WS2812B - all green instead of off

Vizis, 2019-09-07:
Hey 
Im trying to create infinity mirror clock . It't 1st time im using FastLED lib. 
Here is my code: https://pastebin.com/eBDeKn49
and the issue is that all of not used pixels are green. When i set rgb color it looks like color starts from green not 'black'

GianniDPC, 2019-09-08:
```
#define COLOR_ORDER GRB
#define LED_TYPE    WS2812B

```

`  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);`

You need to define the color order as GRB instead of the default RGB.

Vizis, 2019-09-08:
Ok so i repalced my 
 `FastLED.addLeds<NEOPIXEL, LEDStripPin>(leds, numLEDs);`
with ur 
`  FastLED.addLeds<LED_TYPE,LEDStripPin,COLOR_ORDER>(leds, numLEDs).setCorrection(TypicalLEDStrip);`
with
`#define COLOR_ORDER GRB
#define LED_TYPE WS2812B`
of all combionation of RGB (RGB, RBG, BRG, BGR etc)
and allways non used leds was green
(leds strip seems ok because i tested it on "Adafruit_NeoPixel" with sample raibow + effects test"

GianniDPC, 2019-09-08:
> Ok so i repalced my
> `FastLED.addLeds<NEOPIXEL, LEDStripPin>(leds, numLEDs);`
> with ur
> ` FastLED.addLeds<LED_TYPE,LEDStripPin,COLOR_ORDER>(leds, numLEDs).setCorrection(TypicalLEDStrip);`
> with
> `#define [...]

Which MCU as you using? I have the same strip and it works fine. I'm using an STM32F103 MCU.

Vizis, 2019-09-08:
It seems my Lib was damaged somehow... after i reinstalled FastLED everything works like it should be..

closed 2019-09-08 by Vizis