FastLED/FastLED #887: Variable reset

GunMoe, 2019-09-13:
[image]

marmilicious, 2019-09-13:
On line 15,  leds[num],  if num is ever larger then 58 then you might be over writing bits off memory on the controller causing bad things.  The value of num should only be a value of 0 to 58.

henrygab, 2019-10-20:
Marmilicious got it right.  

<details><summary>Details about the bug in your code.</summary>

<P>

In C and C++, an array of length `59` is indexed from `0..58`  (also known as zero-based indexing).  However, your output shows that, eventually, at line 10 you executed with `num` set to `59`.  This is one index too large for the array you allocated.

As a result, when the process uses the value `59` for the array index at line 11, it is referencing the memory **_immediately following_** the `leds` array.

What's in that memory location?  `num` is declared as a static variable immediately before `leds`.  Compilers may (_commonly?_) put static variables next to each other, essentially in reverse order of their declaration.  Thus, the `leds` array may be immediately after the memory location for `num`.  
</details>

How would you fix this bug?

Delete line 18, and replace with code that does the following:
1. If `num >= NUM_LEDS`, then set `num` to `NUM_LEDS` - 1, and set `moe` to `-1`.
2. if `num <= 0`, then set `num` to zero and set `moe` to `+1`.

Hope that helps.