FastLED/FastLED #890: Cannot specify SPI data rate on Teensy 4

woundedlion, 2019-09-18:
The SPI_DATA_RATE template parameter to addLeds() is declared as a uint8_t. On Teensy 4, DATA_RATE_MHZ and DATA_RATE_KHZ are defined to pass the raw clock rate in bps to the Controller class, yielding values larger than 255. The value is truncated resulting in an incorrect/unusable SPI clock rate. I fixed this by redefining as uint32_t.

embedded-creations, 2019-09-20:
Thanks for your fix, I tried it and it worked for me with setting the clock rate for `APA102` LEDs.  I changed `SPI_DATA_RATE` here:

https://github.com/FastLED/FastLED/blob/master/FastLED.h#L223

woundedlion, 2019-09-20:
Sure, posted a PR here: https://github.com/FastLED/FastLED/pull/891