FastLED/FastLED #895: Interrupt problems... can´t solve to my needs

ediciuStsaF, 2019-09-30:
Got some problems using FastLED with RCSwitch library. I´m nearly beginner and stuck in my program. If i make some delay(50); in every void, i´m able to switch the programs of FastLED. But then the "fire" is much to slow to look good.... 
maybe there´s some hint how to get it done.
`#define FASTLED_INTERRUPT_RETRY_COUNT 1
#include <FastLED.h>

#define NUM_LEDS 40
//#define NUM_LEDS_STICK NUM_LEDS/20
#define DATA_PIN 6
#define BRIGHTNESS  200
#define FRAMES_PER_SECOND 60
uint8_t gHue1 = 0;
uint8_t gHue2 = 255;
uint8_t gHueDelta = 3;

bool gReverseDirection = false;
CRGB leds[NUM_LEDS];

#include <RCSwitch.h>

int j = 40;
int Prog = 0;
int i = 0;
int value = 0;
RCSwitch mySwitch = RCSwitch();
CRGBPalette16 gPal;
void setup()
{
  Serial.begin(9600);
  mySwitch.enableReceive(0);  // Empfänger ist an Interrupt-Pin "D2"
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
  i = 0; j = 39;
  FastLED.setBrightness( BRIGHTNESS );
  gPal = HeatColors_p;
  for (i = 0; i <= 39; i++) {
    leds[i] = CRGB::Black;
    FastLED.show();
  }
}
void loop() {

  if (mySwitch.available()) // Wenn ein Code Empfangen wird...
  { Prog = 0;
    int value = mySwitch.getReceivedValue(); // Empfangene Daten werden unter der Variable "value" gespeichert.
    Serial.print(mySwitch.available()); Serial.print("  "); Serial.println(value);
    if (value == -10816 || value == 14144960)
    {
      Prog = 1;
    }
    if (value == -10960 || value == 14144816)
    {
      Prog = 2;
    }

    if (value == -10768 || value == 14145008)
    {
      Prog = 3;
    }
    if (value == -10996 || value == 14144780)
    {
      Prog = 4;
    }
    if (value == -10804 || value == 14144972)
    {
      Prog = 5;
    }
    if (value == -10948 || value == 14144828)
    {
      Prog = 6;
    }
    if (value == -10756 || value == 141450209)
    {
      Prog = 7;
    }
    if (value == -11005 || value == 14144771)
    {
      Prog = 8;
    }
    mySwitch.resetAvailable();  // Hier wird der Empfänger "resettet"
  }
  if (Prog == 1) {

    one();
  }

  if (Prog == 2) {

    two();

  }
  if (Prog == 3) {

    three();

  }
  if (Prog == 4) {

    four();
  }

  if (Prog == 5) {
    five();
  }

  if (Prog == 6) {

    six();
  }
  if (Prog == 7) {
    seven();
  }
  if (Prog == 8) {

    eight();

  }
  EVERY_N_MILLISECONDS( 20 ) {
    gHue1++;
  }

}

void one() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {

    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Red;
      leds[i] = CRGB::Red;
      FastLED.show();
      j--;
    }

  }
}

void two() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Olive;
      leds[i] = CRGB::Olive;
      FastLED.show();
      j--;
    }
  }
}

void three() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Green;
      leds[i] = CRGB::Green;
      FastLED.show();
      j--;
    }
  }
}

void four() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    fill_rainbow( leds, NUM_LEDS / 2, gHue1, 3);
    FastLED.show();
  }
}

void five() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Yellow;
      leds[i] = CRGB::Yellow;
      FastLED.show();
      j--;
    }
  }
}
void six() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Blue;
      leds[i] = CRGB::Blue;
      FastLED.show();
      j--;
    }
  }
}
void seven() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    i = 0; j = 39;
    for (i = 0; i <= 19; i++) {
      leds[j] = CRGB::Purple;
      leds[i] = CRGB::Purple;
      FastLED.show();
      j--;
    }
  }
}
void eight() {
  if (mySwitch.getReceivedValue() == 1)

  { Prog = 0;
  } else {
    random16_add_entropy( random());
    Fire2012WithPalette(); // run simulation frame, using palette colors

    FastLED.show(); // display this frame
    FastLED.delay(1000 / FRAMES_PER_SECOND);
  }
}

#define COOLING  130
#define SPARKING 80

void Fire2012WithPalette()
{
  // Array of temperature readings at each simulation cell
  static byte heat[NUM_LEDS];

  // Step 1.  Cool down every cell a little
  for ( int i = 0; i < NUM_LEDS; i++) {
    heat[i] = qsub8( heat[i],  random8(0, ((COOLING * 10) / NUM_LEDS) + 2));
  }

  // Step 2.  Heat from each cell drifts 'up' and diffuses a little
  for ( int k = NUM_LEDS - 1; k >= 2; k--) {
    heat[k] = (heat[k - 1] + heat[k - 2] + heat[k - 2] ) / 3;
  }

  // Step 3.  Randomly ignite new 'sparks' of heat near the bottom
  if ( random8() < SPARKING ) {
    int y = random8(7);
    heat[y] = qadd8( heat[y], random8(160, 255) );
  }

  // Step 4.  Map from heat cells to LED colors
  for ( int j = 0; j < NUM_LEDS; j++) {
    // Scale the heat value from 0-255 down to 0-240
    // for best results with color palettes.
    byte colorindex = scale8( heat[j], 240);
    CRGB color = ColorFromPalette( gPal, colorindex);
    int pixelnumber;
    if ( gReverseDirection ) {
      pixelnumber = (NUM_LEDS - 1) - j;
    } else {
      pixelnumber = j;
    }
    leds[pixelnumber] = color;
  }
}`

marmilicious, 2019-09-30:
Have you seen this wiki page?
https://github.com/FastLED/FastLED/wiki/Interrupt-problems

ediciuStsaF, 2019-10-01:
Yes, tried around bit without success