FastLED/FastLED #896: FastLED conflicts with analogWrite on ESP8266?

LucianoPez, 2019-09-30:
I've found a weird issue that I cant understand.
Here is a simple code that flash all pixel leds ON and OFF every 250 milliseconds. It works OK until I use the analogWrite function. If I uncomment the analogWrite line, I start to see some individual leds randomly changing color. I've tried different DATA_PIN and different pins on the analogWrite() but still the same. I've tried two different ESP8266 boards (wemos and NodeMcu) and still the same issue.

```
#include <FastLED.h>

#define DATA_PIN     05
#define CANT    28
#define BRIGHTNESS  150

CRGB leds[CANT];
unsigned long syncLastFlash = 0;
boolean lightOn = true;

void setup()
{
  FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, CANT).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(BRIGHTNESS);
  //analogWrite(13, 100);  // This line cause some LEDs to randomly change color
}

void loop()
{
  if (millis() - syncLastFlash > 250)
  {
    syncLastFlash = millis();
    lightOn = !lightOn;

    if (lightOn == true)
    {
      for (byte j = 0; j < CANT; j++)
        leds[j] = CRGB(0, 0, 200);
      FastLED.show();
    }

    else
    {
      for (byte j = 0; j < CANT; j++)
        leds[j] = CRGB(0, 0, 0);
      FastLED.show();
    }
  }
}
```

seymar, 2019-10-10:
What is the state of your electronics?
i.e. If you are driving a motor with that `analogWrite` it could suddenly draw a lot of current.

notrub, 2019-11-30:
I have the same problem.  
I have a string of LEDs on one pin, they have been working great for a while.  I recently added an analogWrite command to control another light through a MOSFET, but if I turn it on my original LEDs get all messed up.  
Both work fine independently, and the analogWrite works fine in conjunction with the LEDs, but the addressable LEDs are not happy with the analogWrite.

shazman-visuals, 2020-01-31:
To my understanding this is a ESP8266 HW issue. Switching to ESP32 helped me.

steadramon, 2020-08-15:
Same here using analogWrite with ESP8266. Found using a delay between .show and analogWrite can help but non fully.

EP32 on order