FastLED/FastLED #901: Why does FastLED.delay contain a "show" function call?

dpinvidic, 2019-10-12:
While writing some simple code, I found that the LED's were constantly updating even after I commented out the FastLED.show call.  It turns out that calling the FastLED.delay function also does a show.
Sorry if this is a dumb question.....but can someone explain/

Regards,
Dan

void CFastLED::delay(unsigned long ms) {
	unsigned long start = millis();
        do {
#ifndef FASTLED_ACCURATE_CLOCK
		// make sure to allow at least one ms to pass to ensure the clock moves
		// forward
		::delay(1);
#endif
		show();
		yield();
	}
	while((millis()-start) < ms);
}

marmilicious, 2019-10-13:
Check out this page:
https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering

dpinvidic, 2019-10-13:
Thanks...That was a big help.
Cheers....

Dan

dpinvidic, 2019-10-13:
That Article explains a lot....

closed 2019-10-13 by dpinvidic