FastLED/FastLED #903: Possible incompatibility using ESP32 and WS2813 light strips

abqmichael, 2019-10-18:
I have been using FastLED for about 2 years and think it's fantastic. My current project is to make WiFi controllable multi-LED lights using ESP32 + FastLED + WS2812B lights. That is going well except for recently, when I tried using WS2813 lights in place of the WS2812 lights I had been using. I cannot get the lights to respond correctly. Instead the lights flicker when sent any command.

In debugging the problem, I wired in connectors for my WS2812B lights so that I could run either one, the other, or both types of strings at once. (The strips plug into the same controller pin so when I run both, they should just run the same light pattern on each.) Additionally, to see whether the problem was with the controller, I wired this up with two different ESP32s: A HiLetgo ESP32 OLED WiFi Kit ESP-32 and a KeeYees ESP32 ESP-32S Development Board. Finally, to make sure this wasn't a soldering problem, I tested my light strips using a different application on an Android Nano board and had no problems there.

In searching for solutions, I came across a github repository (that had re-posted the FastLED code) that had a demoreel written for ESP32. This demo let the user force which core the FastLED library would run on to, I presume, minimize possible cross-library timing interference. The demoreel worked, but didn't make my problem go away.

Have others tested WS2813 lights on ESP32 boards? Can you please comment on success or incompatibilities? Thanks.

kokospalme, 2020-03-24:
I have the same problem with an ESP32 Wrover and about 100px of WS2812. The LEDs are flickering all the time, even when I'm sending only a single color for all pixels

abqmichael, 2020-03-24:
I haven’t had any problems with the WS2812 and ESP32. My largest installation is a Chinley 3m real of 150 pixels which I’ve been running for a month. I’ve also run similar length strings of Altove LEDs in development and have had one string of about 120 LEDs running for several months. I have tested with HiLetgo ESP32 WiFi Kit with Blue OLED Display, KeeYees ESP32 ESP-32S and with Espressif ESP32 development boards. The WS2813s are a non-start.

One thing that’s important is to be sure your connections are solid. I’ve run into problems several times during the breadboard phase thinking I had a software problem but finding it was loose wires. This was especially important when I started using Adafruit Accessories Quad Level-Shifters (Adafruit 1787/Mouser 485-1787), https://www.adafruit.com/product/1787, AKA SN54AHCT125, SN74AHCT125 QUADRUPLE BUS BUFFER GATES WITH 3-STATE OUTPUT).

samguyer, 2020-05-25:
@abqmichael Have you been able to make progress on this issue? We are back working on FastLED more actively, so I'd like to start fixing outstanding problems.

abqmichael, 2020-05-25:
@samguyer. I have pretty much given up on the idea of using WS2813 LEDs. The benefit seems pretty insignificant and the hassle was really high. The main benefit being that they could possibly survive certain types of damage to individual lights. The hassle being that they minimally worked on my Arduino Nano and I totally couldn't get them to work with any ESP32 at all. The debugging got too tedious.

It would be interesting, however, to try out any updates to the library that reportedly got them to work. I still have some of those lights and have a use for them. So I'll watch for updates listed as "fixed WS2813 incompatibility."

samguyer, 2020-05-26:
Hmmm. The two kinds of strips use the exact same timing, so they should be
identical from the software standpoint. Are you using a level shifter to
account for the voltage difference?

abqmichael, 2020-05-26:
Yes. According to the product descriptions, they should use the exact same timing and be compatible with the exact same electronics. As noted, I've tried using them with my Arduino Nano, which works at 5V and doesn't need level converters as well as my ESP32 boards, which operate at 3V. I've gone through several circuit designs and can't remember all the variations I tested them against. I recall that the Nano one used a resistor between the signal wire and the first LED. I did level shifting two different ways with the ESP32. The first was a Bi-directional level shifter from sparkfun (https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide). The second was with an Adafruit Accessories Quad Level-Shifter model 1787 (Mouser 485-1787), https://www.adafruit.com/product/1787. I like the later.

With the Nano, my lights would mostly work but would flicker and produce poor lighting. I thought it was a power supply and went through a lot of testing around the circuit but couldn't make that go away. As to the ESP32s, I couldn't even get a go. In testing, I built some cables so that I could go back and forth between WS2812B and the WS2813 light strips for both my nano and ESP32 setup. Lots and lots of testing.

It's possible that my lights were just bad. I bought mine from Amazon as "ALITOVE 16.4ft 300 Pixels WS2813 Upgraded WS2812B Individually Addressable RGB LED Flexible Strip Light 5050 SMD Dual Signal Wires Waterproof IP67 White PCB 5V DC. It could be that the lights aren't actually the same. They obviously have some different electronics as the WS2813 lights have to read from two different data lines, select one, and then process it. I don't know how they do that but it may be introducing something into the circuit. 

Have you had success with the WS2813s?

abqmichael, 2026-08-17:
I see I forgot to mention the advantage.

The advantage for programming with event systems like this is that the code is (once you get the hang of it) easier to debug. More importantly the code scales better. Writing with a basic event loop is OK if all you want to do is control a set of LEDs. When you add a communications interface (like MQTT or BlueTooth), you start having to take care of a lot of things in that little loop. 

Most people think that code complexity scales with the number of features. This is true if your features and code are independent. However, programming in a single polling loop, like what is taught in typical embedded system tutorials, can easily result in coupled codes. Coupled here means you change something in one place and it affects what's happening in another. With two coupled features, solving issues has you visiting the first feature's solution, then investigating and possibly fixing things for feature 2 then revisiting solution 1 to make sure it works. A potential 4x problem growth for each change. My experience is that code complexity can continue to scale by N^2 or worse. Pretty soon, your work is all about fixing cross-dependencies.

Things like asynchio or Swift helps you decouple these things so that your coding job scales closer to the N that you might desire.

zackees, 2026-08-18:
Michael — this is a fair critique and I'd rather answer it with code than argue with it. FastLED has this now: `fl::task`, shipping since **3.10.4**. It's the same model you're describing — small callbacks that do one thing and hand control back, driven by a loop that owns the timing.

Here's the direct translation of the blinky from the docs page you linked (`blink(led1, 700)` / `blink(led2, 400)`):

```cpp
#include "FastLED.h"
#include "fl/task/task.h"

#define DATA_PIN 5
#define NUM_LEDS 60
CRGB leds[NUM_LEDS];

void setup() {
    FastLED.addLeds<WS2813, DATA_PIN, GRB>(leds, NUM_LEDS);

    // asyncio.create_task(blink(led1, 700))
    fl::task::every_ms(700).then([] {
        static bool on = false;
        on = !on;
        leds[0] = on ? CRGB::Red : CRGB::Black;
    });

    // asyncio.create_task(blink(led2, 400))
    fl::task::every_ms(400).then([] {
        static bool on = false;
        on = !on;
        leds[1] = on ? CRGB::Blue : CRGB::Black;
    });

    // one task owns the hardware push — nothing else calls show()
    fl::task::at_framerate(60).then([] { FastLED.show(); });
}

void loop() {
    fl::task::run();   // this is asyncio.run() — the entire loop body
}
```

That is your description almost line for line: a callback that knows whether the light is on or off, flips it, and ends. `.then()` registers the task — there's no separate "add to the scheduler" step. `fl::task::run()` **is** the do-forever loop you described: look at the clock, run whatever is due, return. `every_ms(n)` and `at_framerate(fps)` are the two recurring flavors. One caveat worth stating plainly: `run()` in `loop()` is mandatory, nothing pumps the scheduler behind your back.

Now the point from your last comment, which I think is the actual one:

> programming in a single polling loop, like what is taught in typical embedded system tutorials, can easily result in coupled codes

Agreed, and the fix isn't the event loop for its own sake — it's that **adding MQTT doesn't mean editing the LED code.** In the sketch above, a new feature is a new task:

```cpp
fl::task::every_ms(50).then([] { mqtt.poll(); });
```

The blink tasks don't change. Nothing is rethreaded. That's the property you lose in a hand-rolled `loop()` full of interleaved `millis()` checks, where each new feature has to be woven through everything already there — which is exactly the coupling you're describing.

The one thing a plain callback scheduler *doesn't* give you is sequential async: "do this, wait for the answer, then do that," without shredding it into a state machine. FastLED has `await` for that, running on an OS-level coroutine:

```cpp
#include "fl/net/http.h"
#include "fl/task/executor.h"

fl::task::coroutine({
    .func = [] {
        auto r = fl::task::await(fl::net::http::fetch_get("http://my-server/color"));
        if (r.ok()) {
            apply_color(r.value().text());        // reads top to bottom
        } else {
            FL_WARN("fetch failed: " << r.error().message);
        }
    },
    .name = "color-fetch"
});
```

That blocks the coroutine, not the LEDs — the animation tasks keep running at 60 fps while it waits on the network. Coroutines and `await` are ESP32 and host builds (you're on ESP32, so that's you); the `every_ms` / `at_framerate` scheduler runs everywhere, down to AVR.

And to return your question — what you can listen for:

- **UI controls** — `UIButton::onClicked()`, `UISlider::onChanged()`, `UICheckbox::onChanged()`, `UIDropdown::onChanged()`
- **Frame lifecycle** — subclass `EngineEvents::Listener` for `onBeginFrame()`, `onEndShowLeds()`, `onEndFrame()`
- **Promises** — `.then()` / `.catch_()` on anything async; `fetch_get()` returns one
- **Network** — `fl::net::ble` for BLE transport and status in 3.10.4; the non-blocking WiFi manager (`fl::net::wifi::connectSta()`, poll `status()`) is on master and lands in the next release

The runnable version of the first sketch is `examples/Async` in the library.

The RTOS ergonomics gap you left over was real, and it's a large part of what I've been working on. If those WS2813 strips are still in a drawer, `addLeds<WS2813, ...>` plus the scheduler above is a pretty different experience than the one you walked away from.

abqmichael, 2026-08-19:
Thank you for the correction. I had forgotten that I had tried some of the asynchronous IO features available in the development system prior to switching to MicroPython. My memory is hazy as it’s been several years. All I really remember clearly was that I spent several months debugging using that system and never could stomp down a crash that kept occurring. Then I switched to MicroPython and had none of those sorts of problems. Maybe it was just that the platform of the time was far more stable. Maybe it was easier to develop stable codes. Maybe MicroPython was just better tuned to me.Michael J.