FastLED/FastLED #911: Teensy 4 Compile error SPI.h No such file or directory

seymar, 2019-10-25:
I am trying to build FastLED using PlatformIO for Teensy 4 and keep getting the following errors: related to `SPI.h`
`FastLED_ID126/platforms/arm/mxrt1062/fastspi_arm_mxrt1062.h:7:17: fatal error: SPI.h: No such file or directory`

FastLED 3.3.2
Teensy 4.4.0 Platform
framework-arduinoteensy 1.147.0 (1.47)

seymar, 2019-10-25:
I added the following to `./.pio/libdeps/FastLED_ID126/library.json`
```json
"dependencies":[
     {
         "name": "SPI",
         "frameworks": "arduino"
     }
]
```

And now it works, but I have a feeling this is not the correct way to fix this issue.

jspolsky, 2022-05-21:
The platformIO library dependency finder usually finds dependencies; since FastLED depends on SPI the ldf should have figured it out for you. However, it has a "quirk" that it will not discover dependencies recursively, e.g., it will not discover that FastLED depends on SPI, unless you include the header (in this case FastLED.h).

So just #include <FastLED.h> in your main.cpp and you should be good to go.

seymar, 2022-05-21:
@jspolsky Thanks for your solution. However since I don't have this issue nor the environment to test it in anymore I think this issue best be closed.

closed 2022-05-21 by seymar