FastLED/FastLED #912: teensy 4.0 only parallel output works, garbage output when FastLED.delay() is used

apleschu, 2019-10-25:
With a teensy 4.0 only parallel output works. When compiling with without the parallel output it compiles fine, but the output is garbage.

FastLED.delay() also creates garbage output. The same code changed to delay() works.

marmilicious, 2019-10-27:
@apleschu  Please put your code on pastebin.com or gist.github.com and share a link to the code so others might be able to better help you.

apleschu, 2019-10-29:
This very simple code works: Please note that this is for parallel output. when I change this for non parallel output, the same code on the same HW does not work. Also if the delay() calls are changed to FastLED.delay() again, the code does not work. This is the most simple I could make the example to show the problem. 

The hardware in question is a teensy4.0 and a strip of 2812B attached. No level shifters nothing else that could create a problem.

---

```
#include <FastLED.h>
#define NUM_LEDS 1
#define DATA_PIN 10

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
     FastLED.addLeds<1,WS2812B, DATA_PIN,GRB>(leds, NUM_LEDS);
     FastLED.setBrightness(10);
}

void loop() { 
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
 
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

marmilicious, 2019-11-01:
@apleschu  Have a look at this post.
https://www.reddit.com/r/FastLED/comments/d9z2os/teensy_40_and_fastled_along_with_ws2811_ws2812_ws/

apleschu, 2019-11-01:
@marmilicious I have seen this post, thats why I know that parallel output does work. Not the best for code portability, and still a bug in my book, but OK. The problem now starts if you have many strips of different length (as I have in the project I am working on) Parallel output assumes that you  have a number of strips of the same length. 

In addition that does not deal with the problem of the FastLED.delay()

MartyMacGyver, 2020-05-25:
I found the problem:

getMaxRefreshRate() is set to zero by default in `controller.h`. The individual chipset controllers override this to their value of choice (typically 400, sometimes 800). This in turn influences the parent `CFastLED` value of `m_nMinMicros`.

If `m_nMinMicros` is zero, things go poorly (and other side-effects occur - I spent a while today trying to figure out why no more than up to the first three LEDs would work *per strand* if the serial monitor was open.))

A workaround is to explicitly call `FastLED.setMaxRefreshRate(400);` (or less) after addLeds().

I think I know how to fix this, though I'd also like to fix the fact that `block_clockless_arm_mxrt1062.h` to `clockless_block_arm_mxrt1062.h` for consistency so I'll be working on two PRs at least.

MartyMacGyver, 2020-05-25:
Well, there's a snag. This works for shorter strips, but the longer the strip, the more you need to reduce the max refresh rate (e.g., at 400Hz a dozen LEDs work, but if you crank that up to 128 it takes a lower refresh rate.

So I'll keep digging.

closed 2020-05-25 by kriegsman

Beefy-Swain, 2020-09-14:
Not sure why this was closed, it is still an issue on all counts.

PR #1021 (2020-05-25): Ensure getMaxRefreshRate() returns a non-zero default for Teensy 4.x
Fixes #912
commit 60c05c3a (2020-05-25): Ensure getMaxRefreshRate() returns a non-zero default for Teensy 4.x

Fixes #912