FastLED/FastLED #914: I’d like to volunteer time to help with RGBW support

ssilverman, 2019-10-29:
I’d like to volunteer time to help complete the RGBW refactoring/improvements. I know what happened, and so understand that this project isn’t a priority nor is it likely to be completed any time soon. When you’re ready to share any code, even in an incomplete state, I’d like to help.

kriegsman, 2019-10-29:
Thank you very much. 
I’ve been doing some thinking about all this, and I’m hoping to have a plan coming together by the end of the year, and I’ll definitely be including your offer in that now. 

The families and friends have some things to take care of in November, but I really hope to get things moving after that. 

I absolutely appreciate the offer and also the spirit in which it is intended. Thanks for being patient, and engaged.

nixmeer, 2020-02-10:
Hello :) what's the current status on this topic? :)

samguyer, 2020-05-25:
@nixmeer and @ssilverman We are working on it now! We should have a "phase 1" version fairly soon. What kind of involvement would you like to have?

hrgdavor, 2020-05-30:
Hi,

I have an ESP32 and 6 strips WS2812B 1m x 144  .

I need to consider my options, to return to esp8266 where I have workign code in arduino IDE.

Possibly even more problematic, I am using https://github.com/bbulkow/FastLED-idf fork.

I was hoping to control each strip on separate pin on ESP32.

my C++ knowledge is limited (long time Java developer). I could help with testing then maybe some more than that down the line.

Is there a version that could work at least for some basic color change + white pixel control ?

hrgdavor, 2020-05-31:
Actually there is ESP32 specific and simple RMT code https://github.com/JSchaenzle/ESP32-NeoPixel-WS2812-RMT that was easy to convert to RGBW. I will focus my effort to integrating a HTTP/HTTPS server and adding some rest api there for basic color change operations (no animations) and use JS to calculate RGB values I need.

samguyer, 2020-06-01:
@hrgdavor That sounds like a reasonable solution if you need something right now. The issue in FastLED is less about pumping out the bits. In fact, the ESP32 driver doesn't know or care what the bytes mean than it is sending.

The main challenge is the three-channel color model that pervades the code above the driver. So, for example, the code that does color correction as the bits go out. And the code that does all the cool color transforms and calculations.

If you don't need any of that stuff, you can actually use the current FastLED implementation by allocating a CRGB array that is big enough to hold four values for each LED. Then just turn off color correction. For example, 3 RGBW LEDs would need an array of 4 CRGB values to hold all the information. It's a bit of a hassle because the colors don't line up neatly, but we could probably do something with pointer casts to fix it.

hrgdavor, 2020-06-02:
Yes, I need something right now, so I will focus on that first. The code there is short, so I am starting to play with it. I have managed to get http server running and changing colors via http.  
I have work to do on web interface as well. 
I want to look into how esp32 driver works with multiple RMT channels and if I can get it to work somehow in paralel for 3 strips.... 

I have few more strips of RGBW leds aside form main project (my wife wants me to finish the light fixture already :D) ... so I am interested to try out fastled again after I get the stuff working (basic global color changes and per pixel color setup). It needs to work first, then I can look to add advanced stuff.

ssilverman, 2023-05-29:
@kriegsman I know Dan was talking about his RGBW-included rewrite on his "RGBW/16-bit" branch here: https://github.com/FastLED/FastLED/pull/698#issuecomment-455827652

Do you have need of some integration assistance? Is there any way I can help?

ssilverman, 2023-08-21:
Trying to follow up…
@samguyer @kriegsman

ssilverman, 2023-08-23:
Would it be possible for me to access Daniel’s “RGBW” branch? I’d like to see
Where he was going.

ssilverman, 2023-09-07:
Would it be possible for me to access Daniel’s “RGBW” branch? I’d like to see
Where he was going. (@kriegsman @samguyer) (I've been trying to volunteer to help for at least several years now.)

razvanphp, 2025-04-03:
@zackees can you please give us a short update on the status of the implementation you merged? It's not in the readme, so not sure what to expect/test. Thank you!

zackees, 2025-04-07:
Okay i've added it to the readme. Should be obvious. Let me know how it goes.

closed 2025-04-07 by zackees