FastLED/FastLED #917: issue with Arduino Nano 33 ble sense

hccstem, 2019-11-07:
avr.h:12:10: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

exit status 1
Error compiling for board Arduino Nano 33 BLE.

Biohazard90, 2019-12-04:
I have the same issue, looks like the library doesn't recognize and support the platform for the Nano 33/nRF528x correctly yet.

henrygab, 2020-01-11:
OK, here's the basics to enable a new board for the nRF52840:

1. Clone the repo
2. Add a branch for your commits
3. Edit the file […/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h](https://github.com/FastLED/FastLED/blob/master/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h)

OK, so what does step #3 entail?

<details>
  <summary>Knowledge of all chip pin <==> Arduino pin mappings</summary>

  <p>

For each pin that is usable for a given board, there are up to three (yes, 3x!) pin numbers that could be used, one of which has two forms.  Let me explain and give them names:

**ARDUINO** == This is the standard Arduino numbering, often used with a "D" prefix, such as `D2`.  This is most often used to generate a physical external PCB connection that is compatible with one of the official Arduino branded boards, but these numbers might have NOTHING to do with what is used at the lowest chip level.

**CHIP** aka **PORT/PIN** == the chip-level view of the port.  This can be written in two ways, but it's most commonly written in the form `PX.YY`, such as `P0.16` or `P1.10` or the like.  When written as a single integer (for the nRF52 boards), the integer representation is always `(32*X) + YY`. 

**PCB Label** == Each board can have a manufacturer-defined number listed on the PCB.  This might match the Arduino numbers, the chip numbers, or be whatever the board designer felt like throwing on.

What you need is often found in a table in variant.h / variant.cpp in the BSP.

</details>
<details>
  <summary>Knowledge of unique preprocessor define for the board</summary>

  <p>

This one should be pretty easy.  Just build and look at the build log.  Or, do it "correctly" by looking into the BSP's JSON.  See [relevant lines for the AdaFruit Bluefruit nRF52840 Feather Express](https://github.com/FastLED/FastLED/blob/e559fcd1176c901a7c3d1a8ad48a5a77d5702491/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h#L50-L52) as an example.

</details>
<details>
  <summary>Put it all together...</summary>

  <p>

Use the [RedBear BLE Nano 2](https://github.com/FastLED/FastLED/blob/e559fcd1176c901a7c3d1a8ad48a5a77d5702491/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h#L429-L450) as a fairly simple example.  It defines `ARDUINO_RB_BLE_NANO2`, and the board only exposes ten usable pins.

</details>

Expand the above three sections for details at each step.  Then, as the most critical step, VERIFY EACH AND EVERY PIN, USING THE ARDUINO PIN NUMBERING.  Set the pin as output in `setup()`, and use the `FastPin<x>` template to flip the state similar to the blink example.

<details>
  <summary>Untested pseudo-code</summary>

  <p>

```c++
// untested pseudo-code

// Modify this one define to test each pin, one at a time …
// Use multimeter, scope, or LED (+resistor) to verify the correct pin is being manipulated
#define ARDUINO_PIN_TO_TEST 1u

void setup(void) {
    // similar to pinMode(ARDUINO_PIN_TO_TEST, OUTPUT);
    // but also:
    //     sets high drive mode
    //     disables pin sense level
    //     disables pull-up/pull-down
    //     disconnects input buffering
    FastPin<ARDUINO_PIN_TO_TEST>::setOutput();
}
void loop(void) {
    // equivalent to digitalWrite(ARDUINO_PIN_TO_TEST, HIGH);
    FastPin<ARDUINO_PIN_TO_TEST>::hival();
    delay(1000);
    // equivalent to digitalWrite(ARDUINO_PIN_TO_TEST, LOW);
    FastPin<ARDUINO_PIN_TO_TEST>::loval();
    delay(1000);
}
```
</details>

Hope that helps!

henrygab, 2020-01-11:
Short answer: 
No support for the Nano 33 BLE / Nano 33 BLE Sense is expected in very near future.

Details...
There's more required.  The nrf52 support I wrote presumes the use of FreeRTOS.  The Arduino BSP decided to go with mbedos.  Support for mbedos would need someone to encapsulate the freertos-specific code, abstract it, write the mbed-equivalents, and **_test_** both versions.  This is not a huge effort (compared to writing support from scratch), but would still be a fair amount of work.

Alternatively, if you reprogram it to use the AdaFruit bootloader and hack the Adafruit BSP to work with the board, then you might come up with something like the following (untested) list of pin defines:

```C++
_FL_DEFPIN( 0,  3, 1); // D0  is P1.03
_FL_DEFPIN( 1, 10, 1); // D1  is P1.10
_FL_DEFPIN( 2, 11, 1); // D2  is P1.11
_FL_DEFPIN( 3, 12, 1); // D3  is P1.12
_FL_DEFPIN( 4, 15, 1); // D4  is P1.15
_FL_DEFPIN( 5, 13, 1); // D5  is P1.13
_FL_DEFPIN( 6, 14, 1); // D6  is P1.14
// D7 and D8 have a mismatch between the diagram and the BSP
_FL_DEFPIN( 9, 27, 0); // D9  is P0.27
_FL_DEFPIN(10,  2, 1); // D10 is P1.02
_FL_DEFPIN(11,  1, 1); // D11 is P1.01
_FL_DEFPIN(12,  8, 1); // D12 is P1.08
_FL_DEFPIN(14,  4, 0); // D14 is P0.04
_FL_DEFPIN(15,  5, 0); // D15 is P0.05
_FL_DEFPIN(16, 30, 0); // D16 is P0.30
_FL_DEFPIN(17, 29, 0); // D17 is P0.29
_FL_DEFPIN(20, 28, 0); // D20 is P0.28
_FL_DEFPIN(21,  3, 0); // D21 is P0.03
```