FastLED/FastLED #918: Keil :: cannot open source input file "avr/io.h": No such file or directory

borbier, 2019-11-13:
I've included FastLED library to my project on Keil. (I've planned to use STM32F767ZI for controlling WS821B) The code in main is fine and didn't shown up errors but when I'm compiled the error is occured as image below.
[image]

Are ther any solution or work-around to get it working?

AndKe, 2019-11-20:
are you using STM32-core ?

borbier, 2019-11-21:
> are you using STM32-core ?

Yes, but I just tried other library and set up configuration for DMA and PWM and made it worked.
[Here is my repository](https://github.com/borbier/stm32f7xxxx-ws2812b)

I appreciated your reply anyway.

closed 2019-11-21 by borbier

AndKe, 2019-11-21:
Great, I am using STM32F103C8 , do you think it'll work?