FastLED/FastLED #919: Can't compile for Adafruit NRF52832

etmeister, 2019-11-13:
When attempting to compile, I get the following error:

```

In file included from C:\Users\technode\AppData\Local\Temp\arduino_modified_sketch_880304\FirstLight.ino:6:0:

C:\Users\technode\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002

 #    pragma message "FastLED version 3.003.002"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastled_arm_nrf52.h:6:0,

                 from C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms.h:11,

                 from C:\Users\technode\Documents\Arduino\libraries\FastLED/FastLED.h:52,

                 from C:\Users\technode\AppData\Local\Temp\arduino_modified_sketch_880304\FirstLight.ino:6:

C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h: In static member function 'static constexpr uintptr_t __generated_struct_NRF_P1::r()':

C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:82:16: error: 'NRF_P1_BASE' was not declared in this scope

         return NRF_P1_BASE;

                ^~~~~~~~~~~

C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:82:16: note: suggested alternative: 'NRF_P0_BASE'

         return NRF_P1_BASE;

                ^~~~~~~~~~~

                NRF_P0_BASE

C:\Users\technode\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:83:5: error: body of constexpr function 'static constexpr uintptr_t __generated_struct_NRF_P1::r()' not a return-statement

     }

     ^
```

This is using FastLED 3.3.2, using the provided 'FirstLight' example sketch. I am using an Adafruit Feather NRF52382.  This seems like it may be related to https://github.com/FastLED/FastLED/issues/842, which was closed as fixed.

etmeister, 2019-11-13:
Uncommenting the #define entries in fastpin_arm_nrf52.h solved this for me:
```
#define NRF_P0_BASE   0x50000000UL
#define NRF_P1_BASE   0x50000300UL
```

henrygab, 2020-01-31:
@etmeister -- The last known issue was merged yesterday.  If you pull the latest master, it should compile now without your having to change files locally.  Hope that helps!