FastLED/FastLED #921: Failure to compile on STM32-Core (the official STM board library) avr/io.h: No such file or directory

AndKe, 2019-11-20:
```
n file included from /home/andre/Arduino/projects/libraries/FastLED/led_sysdefs.h:37,Alternatives for avr/io.h: []

ResolveLibrary(avr/io.h)
  -> candidates: []
                 from /home/andre/Arduino/projects/libraries/FastLED/FastLED.h:41,
Multiple libraries were found for "FastLED.h"
 Used: /home/andre/Arduino/projects/libraries/FastLED
                 from /home/andre/Arduino/projects/libraries/FastLED/examples/ColorPalette/ColorPalette.ino:1:
/home/andre/Arduino/projects/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
          ^~~~~~~~~~
compilation terminated.
Using library FastLED at version 3.3.2 in folder: /home/andre/Arduino/projects/libraries/FastLED 
exit status 1
Error compiling for board Generic STM32F1 series.

```

ForsakenHarmony, 2020-01-05:
Having the same problem, board is a nucleo-64 stm32l476rgt6

matthewblackler, 2020-01-26:
Same problem.  Any ideas?

AndKe, 2020-01-26:
If only one competent developer would look into this...

kriegsman, 2020-01-28:
So if you look at led_sysdefs.h on like 27, you'll see place where it tests for the STM32 platform.  
https://github.com/FastLED/FastLED/blob/master/led_sysdefs.h#L27

```
...
#elif defined(STM32F10X_MD) || defined(__STM32F1__)
#include "platforms/arm/stm32/led_sysdefs_arm_stm32.h"
...
```

I bet that if you add the correct constant as a third option to line 27, it'll select STM32 instead of AVR, and then we'll be past this problem.  

Anyone want to give it a try?  I don't know what the right constant is for "STM32-Core", but perhaps someone with this kind of a board can help out here?

AndKe, 2020-01-28:
@kriegsman I have it all hooked up, how can I help/find the right constant?

kriegsman, 2020-01-28:
Well, I just looked at the long 'compile' commands that you posted (thank you!) and it looks like one of them contains this: `-DSTM32F1xx` and also `-DSTM32F103xB`, so maybe we're looking for __STM32F1xx__, or __STM32F103xB__ .  Or maybe a generic __STM32F1__ ? 

So try adding one of those on line 27 in led_sysdefs.h, and see if one of those let's it compile -- and if it compiles, does it work?  

(You could probably do it with "STM32F1xx" without the underscores, but I'm looking for the underscore version that works, I believe, to include back into the library.)

Thanks for the help!

AndKe, 2020-01-29:
it got further with 
#elif defined(STM32F10X_MD) || defined(__STM32F1__) || defined(STM32F103xB) || defined(__STM32F1xx__) || defined(__STM32F103xB__)
so one of them was good :) 

Now there is a lot of errors about "SystemCoreClock"  and a fey other.
https://gist.github.com/AndKe/e5ff2958eb38c110ef0075f1f3e89f20

AndKe, 2020-01-30:
@kriegsman Please see above, what would be the next thing to try?

kriegsman, 2020-01-30:
Thanks for the help, and the verbose output (above).
Questions you could help with: first, see if you can figure out _which_ of those codes made the compilation succeed further?  I.e., was it `__STM32F1xx__` or `__STD32F103xB__` ?  And second, what is the exact name of the 'board' that you are selecting in the Arduino IDE Tools menu?   (And while I'm asking... what board is this, and where did you get it?)

kriegsman, 2020-01-30:
Also, see if you can find out what the value of F_CPU is on your platform.  It looks like maybe it's not defined to be an actual specific constant value for some reason? (but it probably needs to be, for FastLED to get the cycle timing right)

kriegsman, 2020-01-30:
Ah yes, my esteemed colleagues @focalintent and @GianniDPC  our had already tracked this down and was discussing it over in https://github.com/FastLED/FastLED/pull/872 which arose from an issue in the Arduino STM32 support itself, possibly now corrected? https://github.com/stm32duino/Arduino_Core_STM32/pull/613

So it looks like we're all chasing the same issues here.

AndKe, 2020-01-30:
@kriegsman 
defined(STM32F103xB)   did the trick.  
The Board "Generic STM32F1 series"
the "board part number" is "BluePill103C8"
The actual hardware is one of the many "BluePill" boards on ebay.

F_CPU may not be defined, but should be 72Mhz  (can I set it somewhere?)

As for "possibly now corrected" the https://github.com/stm32duino/Arduino_Core_STM32/pull/613 fix was merged for 1.7.0 milestone, now I am on 1.8.0 release, and experiencing the problem..  so ... no.

kriegsman, 2020-01-30:
Thank you for the info about `__STM32F103xB__` and the board info.

You're right that the upstream "fix" doesn't completely _solve_ the problem by itself but I believe that it is now _solvable_, which it was not before.  I'll look into it, but basically I think we need to figure out where and how to `#undef` and then re- `#define` `F_CPU` (to 72000000).

Hackswell, 2020-04-22:
from fastspi_bitbang.h

`	#define MIN_DELAY ((NS(35)>3) ? (NS(35) - 3) : 1)
`
Then in fastled_delay.h:
```
// Macro to convert from nano-seconds to clocks and clocks to nano-seconds
// #define NS(_NS) (_NS / (1000 / (F_CPU / 1000000L)))
#define F_CPU_MHZ (F_CPU / 1000000L)

// #define NS(_NS) ( (_NS * (F_CPU / 1000000L))) / 1000
#define NS(_NS) (((_NS * F_CPU_MHZ) + 999) / 1000)
#define CLKS_TO_MICROS(_CLKS) ((long)(_CLKS)) / (F_CPU / 1000000L)
```

And finally in: STM32/hardware/stm32/1.8.0/cores/arduino/stm32/stm32_def.h: 
`#define F_CPU SystemCoreClock
`
Which SystemCoreClock is a function, not a constant.  CAN this be hard-coded safely, or is the proper usage for STM32 to use the SystemCoreClock() function?  The clock of the STM32 can be changed, correct?   So, might need to be put into a function rather than a macro, no?  Could the value be set once in a static variable in the function after SystemCoreClock() is called, then just return the cached value?

sallevan, 2020-04-22:
> Which SystemCoreClock is a function, not a constant. CAN this be hard-coded safely, or is the
> proper usage for STM32 to use the SystemCoreClock() function?
No and no.

No, it can not be hard-coded safely: @focalintent wrote some time ago that we **will** hard-code it, but software that uses FastLED **should not** attempt to change core frequency.
This is because of how FastLED has been written.

No, we can not use non-constant F_CPU definition because FastLED relies on compile-time calculations involving that constant (specifically, template instantiation).

> The clock of the STM32 can be changed, correct?
Yes, this is correct.

Hackswell, 2020-04-22:
Understood!   
Then on STM32F4, F_CPU should just be set to HSI_VALUE instead, no?

from system_stm32f4xx.c:
```
#if !defined  (HSE_VALUE)
  #define HSE_VALUE    ((uint32_t)25000000) /*!< Default value of the External oscillator in Hz */
#endif /* HSE_VALUE */

#if !defined  (HSI_VALUE)
  #define HSI_VALUE    ((uint32_t)16000000) /*!< Value of the Internal oscillator in Hz*/
#endif /* HSI_VALUE */

...

uint32_t SystemCoreClock = 16000000;
```

Hackswell, 2020-04-23:
Has anyone reviewed GianniDPC's pull requests?  It seems like they've made good progress on the STM32F1xx line of FastLED.  If their pull requests are sufficient, I'd like to piggy-back my work off theirs.   Maybe pull the requests into a STM32 branch?

GianniDPC, 2020-05-05:
It's been some time. But it was working with my PR's. I'm using it on one of BluePills right now to drive  a WS2812 led strip.

Can you try cherry picking commits from my PR?

AndKe, 2020-07-26:
@GianniDPC 
It is a while since, now I get (with FastLED 3.3.3):
```
/home/andre/Arduino/projects/libraries/FastLED/platforms/avr/clockless_trinket.h:6:10: fatal error: avr/interrupt.h: No such file or directory
    6 | #include <avr/interrupt.h> // for cli/se definitions
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

please point me to the related PR's, if any .

closed 2024-10-14 by zackees