FastLED/FastLED #923: ESP32 SPI problems

nikolindima, 2019-11-23:
Add FastLED lib to my project and Adafruit_PN532 nfc module not working

Get this message.
/Documents/Arduino/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

How i can fix it? i don't use SPI with FastLED.

I use
#include <FastLED.h>
#define NUM_LEDS 18
#define DATA_PIN 22
CRGBArray<NUM_LEDS> leds;

void setup() {
{
FastLED.addLeds<WS2813, DATA_PIN, RGB>(leds, NUM_LEDS);
}

closed 2019-11-28 by nikolindima

Ic-ks, 2020-02-03:
Why is this issue closed without any comment? I also get this warning on a new esp32 (https://www.ebay.com/itm/ESP32-Dev-Board-NodeMCU-WROOM32-WiFi-WLAN-Bluetooth-Development/254138283456) with the latest FastLED version (3.3.3)

kriegsman, 2020-02-04:
Thanks for checking in about this. That message isn’t an error- just a notice. We should change it or remove it.

Ic-ks, 2020-02-04:
But using the SPI output would be less computation intensive, right? And this would result in less power consumption. What is the reason that the esp32 cannot run with the SPI output strategy?

kriegsman, 2020-02-05:
The FastLED code for ESP32 uses the ESP32's dedicated "RMT" hardware unit to output the data signal, instead of the "SPI" hardware as is used on (e.g.) AVR microcontrollers.  But FastLED on ESP32 doesn't actually 'manually bitbanged' the data out using just software.  So I think that since we are already using a hardware-assisted data output channel, we should probably just eliminate this message when that's the case.

Ic-ks, 2020-02-06:
Thank you for the detailed explanation :+1:

bbulkow, 2020-09-15:
The warning actually states that SPI led strings - like the APA102 - will default to bitbanging. The RMT hardware can't be used for SPI leds, because it requires two wires of synchronization.