FastLED/FastLED #925: strip unresponsive when using ESP32 Ethernet library, ETH.h

tehuster, 2019-11-24:
Hi,

Fastled is working great on the esp32. However when i initialise the ethernet connection with this library : https://github.com/espressif/arduino-esp32
ETH.begin(); (line 213 main.cpp)

The ledstrip becomes unresponsive. I'm not getting any compile or runtime errors.

My project can be found here:https://github.com/tehuster/EscapeServer/tree/master/ESP32/Names/src

Please help!

samguyer, 2020-05-25:
Hi @tehuster do you still need help with this issue? I recommend moving over to the FastLED Subreddit

tehuster, 2020-05-27:
Hi @samguyer thanks for your reply. Unfortunatly i have not been able to fix this problem. I also posted the question in the subreddit and you and I talked about it. The problem is not with FastLed but with the espressif ethernet library. So you can close this issue. Have a nice day.

closed 2020-05-27 by tehuster