FastLED/FastLED #926: https://github.com/TheAustrian/ESP8266-LED-Websockets

dbgnu76, 2019-11-25:
Need som help with this sketch ,, 
using a esp8266 nodemcu 12E

I have a 300Led RGBW,  
Problem is i can only add 254 Led in the sketch, all above that number will start WDT bootloop.
is it running low on memory ? 

connected with ASUS-RT-AC5300, channel 4
dhcp client start...
ip:192.168.1.164,mask:255.255.255.0,gw:192.168.1.90

Starting with short timeout *WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.164
local ip
192.168.1.164
192.168.1.90
255.255.255.0

HTTP server started
[WS-Server] Server Started.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
Ledtest example
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

Somebody got an idea ? i have tryed many things, nothing help.

/D

marmilicious, 2019-11-30:
FastLED doesn't currently support RGBW pixels. Have you tried your setup with RGB ones?  Are you sure you have adequate power for the whole setup and are you injecting power in one or more places along the strip?  Post a link to your code if you'd like someone to take a look.

dbgnu76, 2019-11-30:
well, the full code is here https://github.com/TheAustrian/ESP8266-LED-Websockets

and of course i have the right power for it ;)
Problem is that it will reboot without the strips connected ,,,and only if i enter more than 254 leds in "Num_Leds"

If i enter lower than 254 leds ,, all is working fine :) with or without strips connected.

i have tryed to search if there is some restriction above 254, didn`t find anything useful yet.
but i think i need somebodys view about it !

dbgnu76, 2019-12-09:
No one who can help out ? my christmas lights is up and running, but i am still missing about 46-leds :(

marmilicious, 2019-12-10:
For general help with sketches please check out the FastLED discussion on Reddit: https://www.reddit.com/r/FastLED/  
(Sorry, probably should have pointed you there to begin with in this case.)

closed 2024-10-14 by zackees