FastLED/FastLED #927: GradientPalettes.h: No such file or directory

SashapounNako, 2019-11-27:
For one of my electronic projects, I use FastLED and Arduino WebSockets  with a Wemos D1 mini, except that when I want to check or even upload it to my Wemos, the error given in the title is displayed. I use everything with their latest version.

SashapounNako, 2019-12-01:
Does anyone have any idea what this problem might be coming from? It's that it blocks me in the rest of my project.

closed 2020-02-24 by SashapounNako

decklez, 2021-04-01:
did you ever find an answer? thanks

SashapounNako, 2021-04-02:
It's been 1 years this is closed... You have to import all the file from the project and not just open with left click just one file.

IoakeimSogiakas, 2022-02-13:
what do you mean exactly

marmilicious, 2022-02-13:
> what do you mean exactly

This is a guess as to what the problem was. If the project has multiple include files then they all need to be available at compile time.  Here's [an example that uses includes](https://github.com/chemdoc77/CD77_FastLED/tree/master/CD77_NeoMatrix_8X8_panel_Fun_with_FastLED).  The main file `CD77_NeoMatrix_8X8_panel_Fun_with_FastLED.ino` includes two other files on line 15 and 16.  If they are not in the folder too then I won't find them and would complain about `The_8x8_Arrays.h` missing, etc.

IoakeimSogiakas, 2022-02-22:
OK thanks

Ioakeim Sogiakas