FastLED/FastLED #933: High memory usage @ STM32F103

baniaque, 2019-12-06:
Hi, 

I'm trying to change from AVR to STM32. I wanted to do this because of more memory and better performance. When it comes to performance, it's great. Unfortunately, more memory was eaten during the compilation process from the Arduino IDE.

Arduino Pro Mini:
Sketch uses 15448 bytes (50%) of program storage space. Maximum is 30720 bytes.
Global variables use 925 bytes (45%) of dynamic memory, leaving 1123 bytes for local variables. Maximum is 2048 bytes.

STM32F103C Blue Pill:
Sketch uses 83520 bytes (63%) of program storage space. Maximum is 131072 bytes.
Global variables use 7128 bytes (34%) of dynamic memory, leaving 13352 bytes for local variables. Maximum is 20480 bytes.

Simple test code, blink Red and Blue, matrix 9x8 pixels. 

Why STM32 compilation use so many memory? 

`#include <FastLED.h>
#define NUM_LEDS 8
#define BRIGHTNESS  50

CRGBArray<NUM_LEDS> linia1;
CRGBArray<NUM_LEDS> linia2;
CRGBArray<NUM_LEDS> linia3;
CRGBArray<NUM_LEDS> linia4;
CRGBArray<NUM_LEDS> linia5;
CRGBArray<NUM_LEDS> linia6;
CRGBArray<NUM_LEDS> linia7;
CRGBArray<NUM_LEDS> linia8;
CRGBArray<NUM_LEDS> linia9;

void setup() {

  FastLED.setBrightness(BRIGHTNESS );
  FastLED.addLeds<NEOPIXEL, PB9>(linia1, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB8>(linia2, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB7>(linia3, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB6>(linia4, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB5>(linia5, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB4>(linia6, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PB3>(linia7, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PA15>(linia8, NUM_LEDS);
  FastLED.addLeds<NEOPIXEL, PA10 >(linia9, NUM_LEDS);

  Serial.begin(115200);

}

void loop() {

  linia1 = CRGB::Red;
  linia2 = CRGB::Red;
  linia3 = CRGB::Red;
  linia4 = CRGB::Red;
  linia5 = CRGB::Red;
  linia6 = CRGB::Red;
  linia7 = CRGB::Red;
  linia8 = CRGB::Red;
  linia9 = CRGB::Red;
  Serial.println("RED");
  FastLED.show();
  delay(500);

  linia1 = CRGB::Blue;
  linia2 = CRGB::Blue;
  linia3 = CRGB::Blue;
  linia4 = CRGB::Blue;
  linia5 = CRGB::Blue;
  linia6 = CRGB::Blue;
  linia7 = CRGB::Blue;
  linia8 = CRGB::Blue;
  linia9 = CRGB::Blue;
  Serial.println("BLUE");
  FastLED.show();
  delay(500);
}`

yesyesuk, 2019-12-06:
I don't know much about the STM32, but as an experiment write a very simple sketch. Maybe Serial.start(115200); in setup() and Serial.print("Hello World!"); in loop(). Then compile for AVR and STM32 and compare the sizes. The "environment" might be a lot larger for STM32. I also noticed that on ESP8266.

baniaque, 2019-12-10:
https://github.com/bblanchon/ArduinoJson/issues/393

Solution form above link:

add --specs=nano.specs
at the very end of line in Documents\Arduino\hardware\Arduino_STM32\STM32F1\platform.txt on the line that starts with "recipe.c.combine.pattern=...."

Sketch uses 36608 bytes (27%) of program storage space. Maximum is 131072 bytes.
Global variables use 13376 bytes (65%) of dynamic memory, leaving 7104 bytes for local variables. Maximum is 20480 bytes.

VS

Sketch uses 93896 bytes (71%) of program storage space. Maximum is 131072 bytes.
Global variables use 17584 bytes (85%) of dynamic memory, leaving 2896 bytes for local variables. Maximum is 20480 bytes.

Still high dynamic memory usage but sketch size is small :)